moononournation/Arduino_GFX #1: My two displays not works with the library

Hans007a, 2019-12-01:
Hello,

I have two displays, which Your library supports the chipsets, but the displays wont to work with it...
I tried it with Your example.

first display: [](http://www.lcdwiki.com/3.5inch_RPi_Display) chipset ILI9486 (works with TFT_eSPI)
second display: [](http://www.lcdwiki.com/MHS-4.0inch_Display-B) chipset ST7796S (not supported by TFT_eSPI)

Can You check why the display not works please?

Thank You!

Hans

moononournation, 2020-01-02:
Hi, sorry for late reply. Can you give me more details? Hardware photo? Where you bought it? Connection details?
For ILI9486 display, can you give me your working TFT_eSPI config?

nodoubtman, 2020-05-22:
Hello. I have the same problem for the ILI9341 driver with ESP32.

My TFT-eSPI setup is:

  #define TFT_MISO 19
  #define TFT_MOSI 23
  #define TFT_SCLK 18
  #define TFT_CS    5  // Chip select control pin
  #define TFT_DC   32  // Data Command control pin
  #define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

and for the arduino-gfx:
#include "SPI.h"
#include "Arduino_HWSPI.h"
#include "Arduino_ESP32SPI.h"
#include "Arduino_SWSPI.h"
#include "Arduino_GFX.h"            // Core graphics library
#include "Arduino_Canvas.h"         // Canvas (framebuffer) library
#include "Arduino_Canvas_Indexed.h" // Indexed Color Canvas (framebuffer) library
#include "Arduino_HX8347C.h"        // Hardware-specific library for HX8347C
#include "Arduino_HX8352C.h"        // Hardware-specific library for HX8352C
#include "Arduino_HX8357B.h"        // Hardware-specific library for HX8357B
#include "Arduino_ILI9225.h"        // Hardware-specific library for ILI9225
#include "Arduino_ILI9341.h"        // Hardware-specific library for ILI9341
#include "Arduino_ILI9481_18bit.h"  // Hardware-specific library for ILI9481
#include "Arduino_ILI9486_18bit.h"  // Hardware-specific library for ILI9486
#include "Arduino_SEPS525.h"        // Hardware-specific library for SEPS525
#include "Arduino_SSD1283A.h"       // Hardware-specific library for SSD1283A
#include "Arduino_SSD1331.h"        // Hardware-specific library for SSD1331
#include "Arduino_SSD1351.h"        // Hardware-specific library for SSD1351
#include "Arduino_ST7735.h"         // Hardware-specific library for ST7735
#include "Arduino_ST7789.h"         // Hardware-specific library for ST7789
#include "Arduino_ST7796.h"         // Hardware-specific library for ST7796

Arduino_ESP32SPI *bus = new Arduino_ESP32SPI(TFT_DC /* DC */, TFT_CS /* CS */, TFT_SCLK /* SCK */, TFT_MOSI /* MOSI */, -1 /* MISO */);
Arduino_ILI9341 *gfx = new Arduino_ILI9341(bus, TFT_RST);

Thanks for your help :)
Marc.

moononournation, 2020-05-23:
My latest commit should fixed it, here are the demo video: https://twitter.com/moononournation/status/1264138340795731968

closed 2020-05-23 by moononournation