moononournation/Arduino_GFX #4: Help with flickering

arkhipenko, 2020-09-11:
Hi, does your library support off-screen canvases?
In other words, can I do this:

``` c++
// In global declarations:
GFXcanvas1 canvas(128, 32); // 128x32 pixel canvas

// In code later:
canvas.println("I like cake");
tft.drawBitmap(x, y, canvas, 128, 32, foreground, background); // Copy to screen
```
as it is done in Adafruit version of GFX?

I am interested in implementing blinking-free redraws...
https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts

Thanks!

Great work supporting 3-wire SPI! I searched high and low before finding your lib. Thanks!

moononournation, 2020-09-11:
The implementation is a little bit different, currently only assume a full screen canvas flush to another gfx:
`
Arduino_ST7789 *output_display = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */);
// 16-bit color Canvas (240x320 resolution only works for ESP32 with PSRAM)
Arduino_Canvas *gfx = new Arduino_Canvas(240 /* width */, 320 /* height */, output_display);
gfx.println("I like cake");
gfx.flush();
`

arkhipenko, 2020-09-13:
So I cannot use this without PSRAM? That's a bummer...

moononournation, 2020-09-14:
If your program is not a memory hunger, you can still use it with 240x320. Or if your program not use more than 256 colors, you can use Arduino_Canvas_Indexed class instead.

arkhipenko, 2020-09-14:
OK. I will give it a try - thank you!

arkhipenko, 2020-09-14:
Works! Thank you!

closed 2020-09-14 by arkhipenko