moononournation/Arduino_GFX #5: arduino ide 1.8.9无法编译arduno gfx 例程.

LuoDan2016, 2020-09-13:
arduino ide 1.8.9无法编译arduino gfx例程
也无法编译rgb565 video例程。
我很喜欢vide例子。

moononournation, 2020-09-13:
what is the error?

LuoDan2016, 2020-09-14:
exit status 1
C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp: In member function 'virtual void Arduino_TFT::draw24bitRGBBitmap(int16_t, int16_t, const uint8_t*, int16_t, int16_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:651:127: error: operation on 'offset' may be undefined [-Werror=sequence-point]

     writeColor(color565(pgm_read_byte(&bitmap[offset++]), pgm_read_byte(&bitmap[offset++]), pgm_read_byte(&bitmap[offset++])));

                                                                                                                               ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:651:127: error: operation on 'offset' may be undefined [-Werror=sequence-point]

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp: In member function 'virtual void Arduino_TFT::draw24bitRGBBitmap(int16_t, int16_t, uint8_t*, int16_t, int16_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:685:79: error: operation on 'offset' may be undefined [-Werror=sequence-point]

     writeColor(color565(bitmap[offset++], bitmap[offset++], bitmap[offset++]));

                                                                               ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:685:79: error: operation on 'offset' may be undefined [-Werror=sequence-point]

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp: In member function 'virtual void Arduino_TFT::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:973:81: warning: 'yo16' may be used uninitialized in this function [-Wmaybe-uninitialized]

                 writeFillRectPreclipped(x + (xo16 + xx) * textsize_x, y + (yo16 + yy) * textsize_y,

                                                                                 ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT.cpp:843:13: warning: 'xo16' may be used uninitialized in this function [-Wmaybe-uninitialized]

     int16_t xo16, yo16;

             ^

cc1plus.exe: some warnings being treated as errors

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT_18bit.cpp: In member function 'virtual void Arduino_TFT_18bit::draw24bitRGBBitmap(int16_t, int16_t, uint8_t*, int16_t, int16_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_TFT_18bit.cpp:330:11: warning: unused variable 'offset' [-Wunused-variable]

   int16_t offset = 0;

           ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp: In member function 'virtual void Arduino_SSD1283A::writeAddrWindow(uint16_t, uint16_t, uint16_t, uint16_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:137:14: warning: 'h3' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(h3);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:129:14: warning: 'h2' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(h2);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:128:14: warning: 'h1' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(h1);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:136:14: warning: 'v3' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(v3);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:133:14: warning: 'v2' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(v2);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SSD1283A.cpp:132:14: warning: 'v1' may be used uninitialized in this function [-Wmaybe-uninitialized]

   _bus->write(v1);

              ^

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SEPS525.cpp: In member function 'virtual void Arduino_SEPS525::setRotation(uint8_t)':

C:\Users\Administrator\Documents\Arduino\libraries\Arduino_GFX\Arduino_SEPS525.cpp:168:11: warning: unused variable 'startline' [-Wunused-variable]

   uint8_t startline = (_rotation < 2) ? HEIGHT : 0;

           ^

为开发板 ESP32 Dev Module 编译时出错。

moononournation, 2020-09-15:
The error message is ad-normal, e.g.
Arduino_TFT.cpp:651:127: error: operation on 'offset' may be undefined [-Werror=sequence-point]
But line 646 already declared:
uint32_t offset = 0;

I am using Arduino IDE 1.8.13 and everything runs well. Please try to download latest Arduino IDE, latest stable Arduino-ESP32 and latest Arduino_GFX. And then try again.

LuoDan2016, 2020-09-26:
你好，可否给出RGB565 video hex固件？

moononournation, 2020-09-26:
This sample code just in demo purpose, you should add your own logic before you can really use it. So first of all, please clean up your computer and install latest software to make it compile success.

closed 2021-02-02 by moononournation