moononournation/Arduino_GFX #6: error when compiling for GC9A01

PumaFPV, 2020-10-02:
Hi, this is awesome. After seeing your video I ordered the same screen (I needed a circle screen for one project). Today I received it but i cant compile the code I always end up with an error... could you please share your exact code for this display? thanks :)

here is the error: 
multiple libraries were found for "Arduino_HWSPI.h"
In file included from C:\Users\nicol\OneDrive\Documents\Arduino\libraries\Arduino_GFX-master\Arduino_ESP32SPI_DMA.cpp:7:0:

 Used: C:\Users\nicol\OneDrive\Documents\Arduino\libraries\Arduino_GFX-master
C:\Users\nicol\OneDrive\Documents\Arduino\libraries\Arduino_GFX-master\Arduino_ESP32SPI_DMA.h:19:19: fatal error: unity.h: No such file or directory

compilation terminated.

exit status 1

PumaFPV, 2020-10-02:
Here is the code i used

/*
   Arduino Watch Lite Version
   You may find full version at: https://github.com/moononournation/ArduinoWatch
*/

/*******************************************************************************
 * Start of Arduino_GFX setting
 ******************************************************************************/
#include "Arduino_HWSPI.h"
//#include "Arduino_ESP32SPI.h"
//#include "Arduino_SWSPI.h"
#include "Arduino_GFX.h"     // Core graphics library
#include "Arduino_Display.h" // Various display driver

#define TFT_CS 15
#define TFT_DC 2
#define TFT_RST 4
#define TFT_BL 32

/*
 * Step 1: Initize one databus for your display
*/

// General software SPI
// Arduino_DataBus *bus = new Arduino_SWSPI(TFT_DC, TFT_CS, 18 /* SCK */, 23 /* MOSI */, -1 /* MISO */);

// General hardware SPI
Arduino_DataBus *bus = new Arduino_HWSPI(TFT_DC, TFT_CS);

// ESP32 hardware SPI, more customizable parameters
// Arduino_DataBus *bus = new Arduino_ESP32SPI(TFT_DC, TFT_CS, 18 /* SCK */, 23 /* MOSI */, -1 /* MISO */, VSPI /* spi_num */);

/*
 * Step 2: Initize one driver for your display
*/

// Canvas (framebuffer)
// Arduino_ST7789 *output_display = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */);
// 16-bit color Canvas (240x320 resolution only works for ESP32 with PSRAM)
// Arduino_Canvas *gfx = new Arduino_Canvas(240 /* width */, 320 /* height */, output_display);
// Indexed color Canvas, mask_level: 0-2, larger mask level mean less color variation but can have faster index mapping
// Arduino_Canvas_Indexed *gfx = new Arduino_Canvas_Indexed(240 /* width */, 320 /* height */, output_display, 0 /* output_x */, 0 /* output_y */, MAXMASKLEVEL /* mask_level */);

// GC9A01 IPS LCD 240x240
Arduino_GC9A01 *gfx = new Arduino_GC9A01(bus, TFT_RST, 0 /* rotation */, true /* IPS */);

// HX8347C IPS LCD 240x320
// Arduino_HX8347C *gfx = new Arduino_HX8347C(bus, TFT_RST, 0 /* rotation */, true /* IPS */);

// HX8352C IPS LCD 240x400
// Arduino_HX8352C *gfx = new Arduino_HX8352C(bus, TFT_RST, 0 /* rotation */, true /* IPS */);

// HX8357B IPS LCD 320x480
// Arduino_HX8357B *gfx = new Arduino_HX8357B(bus, TFT_RST, 0 /* rotation */, true /* IPS */);

// ILI9225 LCD 176x220
// Arduino_ILI9225 *gfx = new Arduino_ILI9225(bus, TFT_RST);

// ILI9341 LCD 240x320
//Arduino_ILI9341 *gfx = new Arduino_ILI9341(bus, TFT_RST);

// ILI9481 LCD 320x480
// Arduino_ILI9481_18bit *gfx = new Arduino_ILI9481_18bit(bus, TFT_RST);

// ILI9486 LCD 320x480
// Arduino_ILI9486_18bit *gfx = new Arduino_ILI9486_18bit(bus, TFT_RST);

// SEPS525 OLED 160x128
[... 362 lines cut ...]
    if (cross_check_second)
    {
        for (int i = 0; i <= sHandLen; i++)
        {
            if ((x == *(cache++)) && (y == *(cache)))
            {
                return;
            }
            cache++;
        }
    }
    if (cross_check_hour)
    {
        cache = cached_points + ((sHandLen + 1) * 2);
        for (int i = 0; i <= hHandLen; i++)
        {
            if ((x == *(cache++)) && (y == *(cache)))
            {
                return;
            }
            cache++;
        }
    }
    gfx->writePixel(x, y, color);
}

PumaFPV, 2020-10-02:
I'm running on an ESP32 btw. Just like your youtube video, similar wiring but not the exact same

moononournation, 2020-10-03:
multiple libraries were found for "Arduino_HWSPI.h" means you have 2 or more versions of Arduino_GFX library folders, please remove all of them and download latest one again.

uvwxy, 2020-10-03:
@moononournation does the latest one still reference unity.h?

I found this import can be removed, as I had the same error:

'''
C:\Users\nicol\OneDrive\Documents\Arduino\libraries\Arduino_GFX-master\Arduino_ESP32SPI_DMA.h:19:19: fatal error: unity.h: No such file or directory
'''

moononournation, 2020-10-03:
unity.h was added to arduino-esp32 on 27 Nov 2018: https://github.com/espressif/arduino-esp32/pull/2108

Please update the arduino-esp32, some forum discussed how to do that: https://esp32.com/viewtopic.php?t=8284

PumaFPV, 2020-10-03:
I only have one arduino_gfx and I also have the latest version of esp32... (1.0.4)

PumaFPV, 2020-10-03:
I do get the same error on platformio and using only arduino_gfx library, so it cant interfer with anything else...

moononournation, 2020-10-04:
I have tried compile your code in above post with Arduino IDE 1.8.13, latest Arduino-esp32 and Arduino-GFX master branch. Every things goes well if I commented out #endif at line 115.

```
Sketch uses 234111 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13596 bytes (4%) of dynamic memory, leaving 314084 bytes for local variables. Maximum is 327680 bytes.
```

PumaFPV, 2020-10-04:
So I just followed uvwxy advice, which was to comment out the #include "unity.h" and it fixed my problem... which isnt a real solution but that will be enought for me... (I had every latest versions of everything) so thats weird

moononournation, 2020-10-04:
I think you should locate why the compiler cannot find "unity.h" that should included in arduino-esp32. Or you will encounter further wired problems while development.

uvwxy, 2020-10-13:
@moononournation  I think you are absolutely right.

I still have one question: for what is the `unity.h`needed in `Arduino_ESP32SPI_DMA.h`?

I know removing it might not be the correct approach. But it still worked. Question is: why?

moononournation, 2020-10-14:
unity.h should only be used while unit test. But now I know it have a side effect that can prevent you using arduino-esp32 before 27 Nov 2018 ;P

closed 2020-10-14 by moononournation