moononournation/Arduino_GFX #8: Teensy 4.1 HWSPI vs. SWSPI Issues

bsom, 2020-10-31:
When I use SWSPI on a Teensy 4.1 and a GC9A01 display, the library works properly. But when I try HWSPI with the same hardware, I just get lines and screen garbage.  What am I missing?

[Interfacing_Arduino_to_ER-TFTM1.28-1.pdf](https://github.com/moononournation/Arduino_GFX/files/5469942/Interfacing_Arduino_to_ER-TFTM1.28-1.pdf)

Code:
```
#include "Arduino_GFX_Library.h"

//PINS
#define TFT_RST 8
#define TFT_DC 9
#define TFT_CS 10
#define TFT_MOSI 11
#define TFT_SCK 13
#define TFT_MISO -1

Arduino_DataBus *bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_SCK, TFT_MOSI, TFT_MISO);
//Arduino_DataBus *bus = new Arduino_HWSPI(TFT_DC, TFT_CS);
Arduino_GC9A01 *gfx = new Arduino_GC9A01(bus, TFT_RST, 0 /*rotation*/, true /*IPS*/);

void setup() {
  gfx->begin();
}

void loop() {
  gfx->fillScreen(BLACK);
  delay(500);
  gfx->fillScreen(RED);
  delay(500);
}
```

moononournation, 2020-11-01:
Sorry, I did not have Teensy 4.x board in hands. Do you know where can I get one? or any compatible board?

moononournation, 2020-11-01:
I have just checked Teensy Teensyduino official library list: https://www.pjrc.com/teensy/td_libs.html
But it has not yet included Teensy 4.x, I will look forward what modification required from their official code sample.

bsom, 2020-11-01:
https://github.com/PaulStoffregen/cores

Paul is behind on updating some of the pages to reflect the Teensy 4.x. The latest libraries can be found at the GitHub link above.

moononournation, 2020-11-09:
The only SPI display I can found is : https://github.com/PaulStoffregen/Adafruit_ST7735/blob/master/Adafruit_ST7735.cpp
And that example is using software SPI. If you can find an SPI display example using hardware SPI with Teensy, I may be can fix it.

bsom, 2020-11-14:
Where in that example does it say it is using SW SPI?  There are references to both HW and SW SPI in that example but it looks like it uses HW SPI in the code:

```
#elif defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__)
	if (_sid == (uint8_t)-1) _sid = 11;
	if (_sclk == (uint8_t)-1) _sclk = 13;
	if ( spi_pin_is_cs(_cs) && spi_pin_is_cs(_rs)
	 && (_sid == 7 || _sid == 11)
	 && (_sclk == 13 || _sclk == 14)
	 && !(_cs ==  2 && _rs == 10) && !(_rs ==  2 && _cs == 10)
	 && !(_cs ==  6 && _rs ==  9) && !(_rs ==  6 && _cs ==  9)
	 && !(_cs == 20 && _rs == 23) && !(_rs == 20 && _cs == 23)
	 && !(_cs == 21 && _rs == 22) && !(_rs == 21 && _cs == 22) ) {
		hwSPI = true;
```

moononournation, 2020-11-14:
you quota from which file?

bsom, 2020-11-14:
From the file you linked above:
https://github.com/PaulStoffregen/Adafruit_ST7735/blob/master/Adafruit_ST7735.cpp

moononournation, 2020-11-14:
please check the line 182 to 189, that is the code for Teensy and it is
software SPI.

moononournation, 2020-12-18:
Just added Teensy 4.x support, here are the demo video: https://youtu.be/dkhLGyGjbfo

closed 2020-12-18 by moononournation