moononournation/Arduino_GFX #10: ERROR: Failed to open " PNG_FILENAME " file for reading

KevinBennett24, 2020-12-10:
hello, I have some issues when I tried to displayed an image like PNG or GIF, I'm using an NodeMCU ESP8266, also I used sketch data upload but doesn't work, I'm hope you have some solution for this. 

Regards.

moononournation, 2020-12-10:
Please use Arduino ESP8266 filesystem uploader to upload the file:
https://github.com/esp8266/arduino-esp8266fs-plugin.git

KevinBennett24, 2020-12-10:
I am using that version (0.5.0) and got the message "SPIFFS Image Uploaded", also I added to the code #include <fs.h> because I got this error "spiffs begin was not declared in this scope".

moononournation, 2020-12-10:
The code #include <FS.h> just checked in.
If you changed the flash size, both program and and filesystem require upload again.

KevinBennett24, 2020-12-10:
also if you are modify the code you could add this too 
#elif defined(ESP8266)
#define TFT_CS D8
#define TFT_DC D1
#define TFT_RST D2
with that the nodemcu esp8266 works

moononournation, 2020-12-10:
Just checked in a smaller GIF file, now it can play on ESP8266.
The pins definition always free to change while actually use.

KevinBennett24, 2020-12-10:
It worked after reinstalled the drivers for filysystems and the esp8266 community with 2.5.2 version, thanks, just like a note SCL pin to D5 and SDA to D7

moononournation, 2020-12-10:
This caused by JPEGDEC changed method signature in new version, but the new version still have some issue on ESP8266 and ESP32:
https://github.com/bitbank2/JPEGDEC/issues/12

KevinBennett24, 2020-12-11:
alright I'm fine with the png thanks so much, now to try to get a bigger gif play but that uint_8 its the problem I will try other ways

closed 2020-12-11 by moononournation