moononournation/Arduino_GFX #12: Seeeduino XIAO compatibility, swSPI works, hwSPI don’t

Noise4chris, 2021-01-10:
Hello, thank you for the amazing work!

I’m trying to use a Seeeduino XIAO with a GC9A01 round display.
It works via swSPI and hwSPI with the library you can find at this link: 

https://www.buydisplay.com/arduino/ER-TFTM1.28-1_Arduino_Tutorial.zip

But I would like to use your library, it works with swSPI, but doesn’t work wit hwSPI.

There is a way to implement the hwSPI functionality? Or something to edit to make it working?

Thank you very much.

Christian

moononournation, 2021-01-10:
Thx for your sample code. Just committed a fix, please take a look.

Noise4chris, 2021-01-10:
Fantastic, now it works in hwSPI too, thank you.

Is a little bit faster, do you know how to set the clock speed? Is it already at max speed or can I set faster clock speed? 

Thank you again.
Christian

moononournation, 2021-01-10:
you can set speed at gfx.begin()

Noise4chris, 2021-01-11:
Ok, I tried but I see no changes, the fillscreen takes always the same time, about half second to fill the screen...
Thanks again
Christian

closed 2021-01-12 by moononournation