moononournation/Arduino_GFX #13: Pease help! I cant compile sketch for esp8266(((

dos1axgod, 2021-01-16:
I cant even compile example sketches. I tried compile for any esp boards and any example sketch and always i get the same error.

C:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_TFT.cpp: In member function 'virtual void Arduino_TFT::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t)':
C:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_TFT.cpp:1010:1: error: a15 cannot be used in asm here
 }
 ^
C:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_TFT.cpp:1010:1: error: a15 cannot be used in asm here
exit status 1
Compilation error for board Wemos D1 mini.

I ised: 
IDE: Arduino IDE 1.8.13
Board: Wemos D1 mini ESP 8266
Display: GC9A01

moononournation, 2021-01-16:
I have just tested with Arduino IDE 1.8.13 and ESP8266 source code, no this error. You may try update the ESP8266 source code and try again.

dos1axgod, 2021-01-16:
> I have just tested with Arduino IDE 1.8.13 and ESP8266 source code, no this error. You may try update the ESP8266 source code and try again.

I install last update the ESP8266 source code, but still not working((( I dont know what to do.

moononournation, 2021-01-19:
`Arduino_TFT.cpp:1010:1: error: a15 cannot be used in asm here` is odd since it is not asm code. Wild guess it may have file corruption, you may try delete the `C:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino` folder and get the library again.

marcmerlin, 2021-01-30:
Hi, thank you for your good lib. Note that this message is an incompatible change in the newer ESP8266 compiler: https://www.reddit.com/r/esp8266/comments/ggapj8/new_compiler_breaks_programs_error_a15_cannot_be/

If you upgrade your esp8266 compiler, you will likely see the same error.

marcmerlin, 2021-01-30:
This makes the compilation problem go away, of course it also breaks what your code is trying to do:
```
--- a/src/Arduino_TFT.cpp
+++ b/src/Arduino_TFT.cpp
@@ -1039,7 +1039,7 @@ void Arduino_TFT::drawChar(int16_t x, int16_t y, unsigned char c, uint16_t color
               {
                 if (!(bit++ & 7))
                 {
-                  bits = pgm_read_byte(&bitmap[bo++]);
+                  //bits = pgm_read_byte(&bitmap[bo++]);
                 }
                 draw_dot = bits & 0x80;
                 bits <<= 1;
@@ -1096,7 +1096,7 @@ void Arduino_TFT::drawChar(int16_t x, int16_t y, unsigned char c, uint16_t color
           {
             if (!(bit++ & 7))
             {
-              bits = pgm_read_byte(&bitmap[bo++]);
+              ///bits = pgm_read_byte(&bitmap[bo++]);
             }
             if (bits & 0x80)
             {
```

But it confirms that pgm_read_byte broke in the newer ESP8266 compiler. Are you able to replace this with new code that doesn't conflict with the compiler?

moononournation, 2021-01-30:
Is it only related to certain OS? I have no this problem in MacOS and I
also use FastLED with no problems.

marcmerlin, 2021-01-30:
more details here: https://github.com/esp8266/Arduino/issues/4572

marcmerlin, 2021-01-30:
If you have the arduino environment, go to board manager, esp8266, and check your version.
I think it's broken with 2.6.x and 2.7.x but it works with 2.5.x and earlier.  The bug I just linked, gives details.
If I downgrade to 2.5.2, it works, but the problem is that it won't work anymore for anyone installing new arduino.
If you do not have an ESP8266 chip, you can still select ESP8266 and try to compile (verify) to see the error

moononournation, 2021-01-31:
I am using Arduino 1.8.13 on MacOS and just run `git pull origin master --recurse-submodule` under `~/Documents/Arduino/hardware/esp8266com/esp8266`.
It have some warning after turned on all compiler warning nut build success:
```
In file included from /Users/moon/Documents/Arduino/libraries/Arduino_GFX/examples/ImgViewerAnimatedGIF/ImgViewerAnimatedGIF.ino:248:
sketch/GifClass.h: In member function 'void GifClass::read_plain_text_ext(gd_GIF*)':
sketch/GifClass.h:282:19: warning: unused variable 'sub_block' [-Wunused-variable]
  282 |             off_t sub_block;
      |                   ^~~~~~~~~
sketch/GifClass.h: In member function 'void GifClass::reset_table(gd_Table*, int32_t)':
sketch/GifClass.h:418:56: warning: narrowing conversion of 'key' from 'int32_t' {aka 'int'} to 'uint8_t' {aka 'unsigned char'} [-Wnarrowing]
  418 |             table->entries[key] = (gd_Entry){1, 0xFFF, key};
      |                                                        ^~~
sketch/GifClass.h: In member function 'int32_t GifClass::read_image_data(gd_GIF*, int, uint8_t*)':
sketch/GifClass.h:502:15: warning: unused variable 'start' [-Wunused-variable]
  502 |         off_t start, end;
      |               ^~~~~
sketch/GifClass.h:502:22: warning: unused variable 'end' [-Wunused-variable]
  502 |         off_t start, end;
      |                      ^~~
sketch/GifClass.h: In member function 'void GifClass::render_frame_rect(gd_GIF*, uint16_t*, uint8_t*)':
sketch/GifClass.h:627:25: warning: unused variable 'color' [-Wunused-variable]
  627 |         uint8_t index, *color;
      |                         ^~~~~
/Users/moon/Documents/Arduino/libraries/Arduino_GFX/examples/ImgViewerAnimatedGIF/ImgViewerAnimatedGIF.ino: In function 'void setup()':
/Users/moon/Documents/Arduino/libraries/Arduino_GFX/examples/ImgViewerAnimatedGIF/ImgViewerAnimatedGIF.ino:357:31: warning: comparison of integer expressions of different signedness: 'long unsigned int' and 'int' [-Wsign-compare]
  357 |             } while (millis() < delay_until);
      |                      ~~~~~~~~~^~~~~~~~~~~~~
In file included from /Users/moon/Documents/Arduino/libraries/Arduino_GFX/examples/ImgViewerAnimatedGIF/ImgViewerAnimatedGIF.ino:248:
sketch/GifClass.h: In member function 'int32_t GifClass::read_image_data(gd_GIF*, int, uint8_t*)':
sketch/GifClass.h:436:41: warning: 'entry.gd_Entry::suffix' may be used uninitialized in this function [-Wmaybe-uninitialized]
  436 |         table->entries[table->nentries] = (gd_Entry){length, prefix, suffix};
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
sketch/GifClass.h:501:18: note: 'entry.gd_Entry::suffix' was declared here
  501 |         gd_Entry entry;
      |                  ^~~~~
sketch/GifClass.h:536:32: warning: 'str_len' may be used uninitialized in this function [-Wmaybe-uninitialized]
  536 |                 ret = add_entry(gif->table, str_len + 1, key, entry.suffix);
      |                       ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
sketch/GifClass.h:533:18: warning: 'table_is_full' may be used uninitialized in this function [-Wmaybe-uninitialized]
  533 |             else if (!table_is_full)
      |                  ^~
/Users/moon/Documents/Arduino/libraries/Arduino_GFX/src/canvas/Arduino_Canvas.cpp: In member function 'virtual void Arduino_Canvas::begin(int32_t)':
/Users/moon/Documents/Arduino/libraries/Arduino_GFX/src/canvas/Arduino_Canvas.cpp:11:12: warning: unused variable 's' [-Wunused-variable]
   11 |     size_t s = _width * _height * 2;
      |            ^
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 288288          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27344   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1484  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1624  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26760 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 318740 bytes (33%) of program storage space. Maximum is 958448 bytes.
Global variables use 29868 bytes (36%) of dynamic memory, leaving 52052 bytes for local variables. Maximum is 81920 bytes.
```

marcmerlin, 2021-01-31:
Thank you for the details.
I think you are correct, the compiler on linux and MacOS may not be the same.

Do you have a directory called hardware/esp8266 (maybe packages/hardware/esp8266 )?
which version do you have in there? do you have package.json ?

Mine looks like this for an old version
```
sauron [mc]$ cat package.json 
{
    "name": "framework-arduinoespressif8266",
    "description": "Arduino Wiring-based Framework (ESP8266 Core)",
    "url": "https://github.com/esp8266/Arduino",
    "version": "2.5.0-dev"
}
```

moononournation, 2021-02-01:
```
{
    "name": "framework-arduinoespressif8266",
    "description": "Arduino Wiring-based Framework (ESP8266 Core)",
    "url": "https://github.com/esp8266/Arduino",
    "version": "3.0.0-dev"
}
```

marcmerlin, 2021-02-02:
Thank you. It looks like you have access to a completely different compiler on MacOS. This is quite confusing.
@dos1axgod is using windows, I'm using linux, and we are both seeing the exact same problem.

The latest in library manager, gives:
```
{
    "name": "framework-arduinoespressif8266",
    "description": "Arduino Wiring-based Framework (ESP8266 Core)",
    "url": "https://github.com/esp8266/Arduino",
    "version": "2.7.4"
}
```
and it does not work.

But I see you have the git version, so I went to https://github.com/esp8266/Arduino , then went to 
/home/merlin/.arduino15/packages/esp8266/hardware/esp8266 , git clone https://github.com/esp8266/Arduino and renamed it to 3.0.0

Sure enough, that one worked, so they fixed the incompatibility they added in 2.6. This is great news. Unfortunately it's not released yet, so you need to use git to get it. It works for me though @dos1axgod  try it too, it should work for you.

@moononournation , now that it builds again on ESP8266, I see a few new warnings:
```
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:16:16: warning: 'len' defined but not used [-Wunused-variable]
 static int16_t len;
                ^~~
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:15:16: warning: 'step' defined but not used [-Wunused-variable]
 static int16_t step;
                ^~~~
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:14:16: warning: 'xs' defined but not used [-Wunused-variable]
 static int16_t xs;
                ^~
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:13:16: warning: 'err' defined but not used [-Wunused-variable]
 static int16_t err;
                ^~~
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:12:16: warning: 'dy' defined but not used [-Wunused-variable]
 static int16_t dy;
                ^~
/home/merlin/Arduino/libraries/Arduino_GFX/src/Arduino_TFT.cpp:11:16: warning: 'dx' defined but not used [-Wunused-variable]
 static int16_t dx;
                ^~
```

moononournation, 2021-02-02:
Ok, OCD again ;>

closed 2021-02-02 by moononournation

moononournation, 2021-04-28:
hope #43 workaround can help you

marcmerlin, 2021-04-28:
oooh, that's a good idea, you basically undo the change in the new ESP8266 compiler suite and restore the old define they used to have.  Nicely done.