moononournation/Arduino_GFX #14: st7789 135x240 cannot display jpg picture completely.

Eddddddddy, 2021-01-16:
The last 7 columns cannot display with a jpg picture. You can easily reproduce the problem with the example named ImgViewerJpeg.ino. I change the code 
`Arduino_ST7789 *gfx = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */, 135 /* width */, 240 /* height */, 53 /* col offset 1 */, 40 /* row offset 1 */, 52 /* col offset 2 */, 40 /* row offset 2 */);`
to
`Arduino_ST7789 *gfx = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */, 136 /* width */, 240 /* height */, 53 /* col offset 1 */, 40 /* row offset 1 */, 52 /* col offset 2 */, 40 /* row offset 2 */);`
the problem was fixed.

I found something.

> Draw pos = 0,0. size = 64 x 8
Draw pos = 64,0. size = 64 x 8
Draw pos = 128,0. size = 8 x 8   <--------it does not work
Draw pos = 0,8. size = 64 x 8
Draw pos = 64,8. size = 64 x 8
Draw pos = 128,8. size = 8 x 8   <--------it does not work
Draw pos = 0,16. size = 64 x 8
Draw pos = 64,16. size = 64 x 8
Draw pos = 128,16. size = 8 x 8  <--------it does not work
Draw pos = 0,24. size = 64 x 8
Draw pos = 64,24. size = 64 x 8
Draw pos = 128,24. size = 8 x 8  ......
Draw pos = 0,32. size = 64 x 8
Draw pos = 64,32. size = 64 x 8
Draw pos = 128,32. size = 8 x 8
Draw pos = 0,40. size = 64 x 8
Draw pos = 64,40. size = 64 x 8
Draw pos = 128,40. size = 8 x 8
Draw pos = 0,48. size = 64 x 8
Draw pos = 64,48. size = 64 x 8
Draw pos = 128,48. size = 8 x 8
Draw pos = 0,56. size = 64 x 8
Draw pos = 64,56. size = 64 x 8
Draw pos = 128,56. size = 8 x 8
Draw pos = 0,64. size = 64 x 8
Draw pos = 64,64. size = 64 x 8
Draw pos = 128,64. size = 8 x 8
Draw pos = 0,72. size = 64 x 8
Draw pos = 64,72. size = 64 x 8
Draw pos = 128,72. size = 8 x 8
Draw pos = 0,80. size = 64 x 8
Draw pos = 64,80. size = 64 x 8
Draw pos = 128,80. size = 8 x 8
Draw pos = 0,88. size = 64 x 8
Draw pos = 64,88. size = 64 x 8
Draw pos = 128,88. size = 8 x 8
Draw pos = 0,96. size = 64 x 8
Draw pos = 64,96. size = 64 x 8
Draw pos = 128,96. size = 8 x 8
Draw pos = 0,104. size = 64 x 8
Draw pos = 64,104. size = 64 x 8
Draw pos = 128,104. size = 8 x 8
Draw pos = 0,112. size = 64 x 8
Draw pos = 64,112. size = 64 x 8
Draw pos = 128,112. size = 8 x 8
Draw pos = 0,120. size = 64 x 8
Draw pos = 64,120. size = 64 x 8
Draw pos = 128,120. size = 8 x 8
Draw pos = 0,128. size = 64 x 8
Draw pos = 64,128. size = 64 x 8
Draw pos = 128,128. size = 8 x 8
Draw pos = 0,136. size = 64 x 8
Draw pos = 64,136. size = 64 x 8
Draw pos = 128,136. size = 8 x 8
Draw pos = 0,144. size = 64 x 8
Draw pos = 64,144. size = 64 x 8
Draw pos = 128,144. size = 8 x 8
Draw pos = 0,152. size = 64 x 8
Draw pos = 64,152. size = 64 x 8
Draw pos = 128,152. size = 8 x 8
Draw pos = 0,160. size = 64 x 8
Draw pos = 64,160. size = 64 x 8
Draw pos = 128,160. size = 8 x 8
Draw pos = 0,168. size = 64 x 8
Draw pos = 64,168. size = 64 x 8
Draw pos = 128,168. size = 8 x 8
Draw pos = 0,176. size = 64 x 8
Draw pos = 64,176. size = 64 x 8
Draw pos = 128,176. size = 8 x 8
Draw pos = 0,184. size = 64 x 8
Draw pos = 64,184. size = 64 x 8
Draw pos = 128,184. size = 8 x 8
Draw pos = 0,192. size = 64 x 8
Draw pos = 64,192. size = 64 x 8
Draw pos = 128,192. size = 8 x 8
Draw pos = 0,200. size = 64 x 8
Draw pos = 64,200. size = 64 x 8
Draw pos = 128,200. size = 8 x 8
Draw pos = 0,208. size = 64 x 8
Draw pos = 64,208. size = 64 x 8
Draw pos = 128,208. size = 8 x 8
Draw pos = 0,216. size = 64 x 8
Draw pos = 64,216. size = 64 x 8
Draw pos = 128,216. size = 8 x 8
Draw pos = 0,224. size = 64 x 8
Draw pos = 64,224. size = 64 x 8
Draw pos = 128,224. size = 8 x 8
Draw pos = 0,232. size = 64 x 8
Draw pos = 64,232. size = 64 x 8
Draw pos = 128,232. size = 8 x 8
Time used: 279

esp32 4m flash without psram.
screen: st7789 135x240.

moononournation, 2021-01-17:
Yes, I know this behavior. The drawBitmap functions simply skip drawing if the drawing block not all inside the display area. The simplest workaround is change the height to 136:
```
Arduino_ST7789 *gfx = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */, 136 /* width */, 240 /* height */, 53 /* col offset 1 */, 40 /* row offset 1 */, 51 /* col offset 2 */, 40 /* row offset 2 */);
```

Eddddddddy, 2021-01-17:
thanks for that.
By the way, is there a method that I could draw the block by 68x8? That is more efficiency, I guess.

moononournation, 2021-01-17:
I have committed some handling on incomplete drawing in jpegDrawCallback(), please pull latest version to take a look.

Eddddddddy, 2021-01-17:
still not work. 
I try to fix the error. 
what I do:

```
static int jpegDrawCallback(JPEGDRAW *pDraw)
{
// Serial.printf("Draw pos = %d,%d. size = %d x %d\n", pDraw->x, pDraw->y, pDraw->iWidth, pDraw->iHeight);
  if (pDraw->x <= gfx->width())
  {
    int16_t w = pDraw->iWidth;    //that should be width, not height
    if ((pDraw->x + w) >= gfx->width())
    {
      w = gfx->width() - pDraw->x;
    }
    Serial.printf("Draw pos = %d,%d. size = %d x %d\n", pDraw->x, pDraw->y, w, pDraw->iHeight);
    gfx->draw16bitBeRGBBitmap(pDraw->x, pDraw->y, pDraw->pPixels, w, pDraw->iHeight);
    return 1;
  }
  return 1;   //if return 0, it will be stop drawing.
}
```

but the the pixels of last serval columns are dislocation. the image is not right. 
[image]

moononournation, 2021-01-18:
Should be fixed now, please pull latest code and take a look.

Eddddddddy, 2021-01-18:
FANTASTIC JOB！
1.0.6 fixed the error.

thanks a lot.

closed 2021-01-18 by Eddddddddy