moononournation/Arduino_GFX #15: WIO WiFi Analyzer requires change for updated RTL8720 firmware

robertpeteuil, 2021-01-17:
People who have upgraded the firmware on their WIO's [RTL8720](https://wiki.seeedstudio.com/Wio-Terminal-Network-Overview/#libraries-installations-erpc) need to change the include on line 17 to the eRPC library:

from:

https://github.com/moononournation/Arduino_GFX/blob/c467cf2af5102ffab15faac8f801c75c749411c4/examples/WioWiFiAnalyzer/WioWiFiAnalyzer.ino#L17

to

```c
#include "rpcWifi.h"
```

moononournation, 2021-01-18:
Yes, WioTerminal is still in developing substantially. But I suffer from below error and cannot test latest version yet:
```
src/port/esp_hardware.c:11:10: fatal error: esp_system.h: No such file or directory
```

moononournation, 2021-01-18:
I am using these library release:
- https://github.com/Seeed-Studio/Seeed_Arduino_FS/releases/tag/v2.0.2
- https://github.com/Seeed-Studio/Seeed_Arduino_SFUD/releases/tag/v2.0.1
- https://github.com/Seeed-Studio/Seeed_Arduino_mbedtls/releases/tag/v2.0.1
- https://github.com/Seeed-Studio/Seeed_Arduino_rpcUnified/releases/tag/v2.1.1
- https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi/releases/tag/v1.0.2

and latest firmware:
- https://github.com/Seeed-Studio/seeed-ambd-firmware/releases/tag/v2.1.1

but still fatal error: esp_system.h: No such file or directory

Which version you can compile success?

robertpeteuil, 2021-01-18:
Yes, its compiling and working fine for me (using the rpc libs).  

I think you problem you're having, may be related to the `mbedtls` library, as I think you need to use the `dev` branch for the WIO.

The Seeed's docs on installing the new network libs [reference the dev branch](https://wiki.seeedstudio.com/Wio-Terminal-Network-Overview/#3-install-the-seeed_arduino_mbedtls).   I noticed the dev branch contains [this commit](https://github.com/Seeed-Studio/Seeed_Arduino_mbedtls/commit/18d813c67a66072059d661a185559206a2914ee0) which removed ESP dependencies.

These are the libraries I'm using (not release tagged):
```yaml
https://github.com/Seeed-Studio/Seeed_Arduino_FS
https://github.com/Seeed-Studio/Seeed_Arduino_SFUD
https://github.com/Seeed-Studio/Seeed_Arduino_mbedtls/archive/dev.zip
https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi
https://github.com/Seeed-Studio/Seeed_Arduino_rpcUnified
https://github.com/Seeed-Studio/Seeed_Arduino_FreeRTOS
https://github.com/moononournation/Arduino_GFX
```

commit 5675b0f2 (2021-01-19): Fix #15
closed 2021-01-19 by moononournation