moononournation/Arduino_GFX #16: Some bugs with SSD1331

tocaedit, 2021-01-25:
By default the screen is unstable and shows artifacts with solid color.

Fixed by editing Arduino_SSD1331.cpp

Change Original:

```
  _bus->sendCommand(SSD1331_PRECHARGEA); // 0x8A
  _bus->sendCommand(0x64);
  _bus->sendCommand(SSD1331_PRECHARGEB); // 0x8B
  _bus->sendCommand(0x78);
  _bus->sendCommand(SSD1331_PRECHARGEA); // 0x8C
  _bus->sendCommand(0x64);
  _bus->sendCommand(SSD1331_PRECHARGELEVEL); // 0xBB
  _bus->sendCommand(0x3A);
  _bus->sendCommand(SSD1331_VCOMH); // 0xBE
```

To Updated:

```
  _bus->sendCommand(SSD1331_PRECHARGEA); // 0x8A
  _bus->sendCommand(0x0);
  _bus->sendCommand(SSD1331_PRECHARGEB); // 0x8B
  _bus->sendCommand(0x0);
  _bus->sendCommand(SSD1331_PRECHARGEC); // 0x8C
  _bus->sendCommand(0x0);
  _bus->sendCommand(SSD1331_PRECHARGELEVEL); // 0xBB
  _bus->sendCommand(0x0);
  _bus->sendCommand(SSD1331_VCOMH); // 0xBE
  _bus->sendCommand(0x0);
```

moononournation, 2021-01-26:
If you comment out the above 5 settings, is it stable?
My SSD1331 tested ok on both original settings and comment out the above 5 settings.

moononournation, 2021-02-02:
The color deep is much reduced after applied your code, so I don't think this is a good option:
[image]
[image]

closed 2021-02-02 by moononournation