moononournation/Arduino_GFX #17: code does not build with preferences/compiler warnings: all

marcmerlin, 2021-01-30:
If you use the arduino environment, can you go to file/preferences/compiler warning: all

Then, build the animatedgifs example and you get those errors.  They may not be coding errors, but they are enough of a potential problem that the compiler treats them as such:
```
/home/merlin/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/home/merlin/arduino/hardware/espressif/esp32/tools/sdk/include/esp-face -I/home/merlin/arduino/hardware/espressif/esp32/tools/sdk/include/fb_gfx -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=5 -I/home/merlin/arduino/hardware/espressif/esp32/cores/esp32 -I/home/merlin/arduino/hardware/espressif/esp32/variants/esp32 -I/home/merlin/Arduino/libraries/Arduino_GFX/src -I/home/merlin/arduino/hardware/espressif/esp32/libraries/SPIFFS/src -I/home/merlin/arduino/hardware/espressif/esp32/libraries/FS/src -I/home/merlin/arduino/hardware/espressif/esp32/libraries/SPI/src /tmp/arduino_build_753374/sketch/ImgViewerAnimatedGIF.ino.cpp -o /tmp/arduino_build_753374/sketch/ImgViewerAnimatedGIF.ino.cpp.o
In file included from /home/merlin/arduino/libraries/Arduino_GFX/examples/ImgViewerAnimatedGIF/ImgViewerAnimatedGIF.ino:248:0:
/tmp/arduino_build_753374/sketch/GifClass.h: In member function 'void GifClass::read_plain_text_ext(gd_GIF*)':
/tmp/arduino_build_753374/sketch/GifClass.h:282:19: warning: unused variable 'sub_block' [-Wunused-variable]
             off_t sub_block;
                   ^
/tmp/arduino_build_753374/sketch/GifClass.h: In member function 'void GifClass::reset_table(gd_Table*, int32_t)':
GifClass.h:418:59: error: narrowing conversion of 'key' from 'int32_t {aka int}' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing]
             table->entries[key] = (gd_Entry){1, 0xFFF, key};
                                                           ^
/tmp/arduino_build_753374/sketch/GifClass.h: In member function 'int32_t GifClass::read_image_data(gd_GIF*, int, uint8_t*)':
/tmp/arduino_build_753374/sketch/GifClass.h:502:15: warning: unused variable 'start' [-Wunused-variable]
         off_t start, end;
               ^
/tmp/arduino_build_753374/sketch/GifClass.h:502:22: warning: unused variable 'end' [-Wunused-variable]
         off_t start, end;
                      ^
/tmp/arduino_build_753374/sketch/GifClass.h: In member function 'void GifClass::render_frame_rect(gd_GIF*, uint16_t*, uint8_t*)':
/tmp/arduino_build_753374/sketch/GifClass.h:627:25: warning: unused variable 'color' [-Wunused-variable]
         uint8_t index, *color;
                         ^
/tmp/arduino_build_753374/sketch/GifClass.h: In member function 'int32_t GifClass::read_image_data(gd_GIF*, int, uint8_t*)':
GifClass.h:436:77: error: 'entry.gd_Entry::suffix' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         table->entries[table->nentries] = (gd_Entry){length, prefix, suffix};
                                                                             ^
/tmp/arduino_build_753374/sketch/GifClass.h:501:18: note: 'entry.gd_Entry::suffix' was declared here
         gd_Entry entry;
                  ^
GifClass.h:536:32: error: 'str_len' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                 ret = add_entry(gif->table, str_len + 1, key, entry.suffix);
                                ^
GifClass.h:533:18: error: 'table_is_full' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             else if (!table_is_full)
                  ^
cc1plus: some warnings being treated as errors
```

marcmerlin, 2021-01-31:
Thanks for your reply in https://github.com/moononournation/Arduino_GFX/issues/13 , I think you probably wanted to paste your results here.
It looks like you are right and the compiler on linux and MacOS are not the same.
Mine says "error: 'table_is_full' may be used uninitialized in this function [-Werror=maybe-uninitialized]"
and yours says "warning: 'table_is_full' may be used uninitialized in this function [-Wmaybe-uninitialized]"

Even if the compiler output is different, all those warnings are probably worth fixing as some could be real errors. Are you able to fix them?
If you'd like help, I can try to fix any compiler warning you're not sure about, feel free to ask and I'll have a look.

moononournation, 2021-01-31:
Thanks for your information, I also have some OCD on eliminate compiler warning, it should already fixed at last commit.

closed 2021-01-31 by moononournation

marcmerlin, 2021-02-02:
haha, thank you for having OCD like me :)

I still see a few warnings left (thanks for fixing the other ones)
```
/home/merlin/Arduino/libraries/Arduino_GFX/src/databus/Arduino_ESP32SPI_DMA.cpp: In member function 'virtual void Arduino_ESP32SPI_DMA::begin(int32_t, int8_t)':
/home/merlin/Arduino/libraries/Arduino_GFX/src/databus/Arduino_ESP32SPI_DMA.cpp:87:3: warning: missing initializer for member 'spi_bus_config_t::flags' [-Wmissing-field-initializers]
   };

   ^
/home/merlin/Arduino/libraries/Arduino_GFX/src/databus/Arduino_ESP32SPI_DMA.cpp:87:3: warning: missing initializer for member 'spi_bus_config_t::intr_flags' [-Wmissing-field-initializers]
/home/merlin/Arduino/libraries/Arduino_GFX/src/databus/Arduino_ESP32SPI_DMA.cpp:101:3: warning: missing initializer for member 'spi_device_interface_config_t::pre_cb' [-Wmissing-field-initializers]
   };

   ^
/home/merlin/Arduino/libraries/Arduino_GFX/src/databus/Arduino_ESP32SPI_DMA.cpp:101:3: warning: missing initializer for member 'spi_device_interface_config_t::post_cb' [-Wmissing-field-initializers]
```