moononournation/Arduino_GFX #18: GIF playback with the STM32F411 BlackPill

JonRobo, 2021-02-01:
I am trying to compile the example for the BlackPill board and am getting this error

error: 'class Arduino_GC9A01' has no member named 'writeIndexedPixels'
  121 |             gfx->writeIndexedPixels(buf, gif->palette->colors, s);
  
Looking through the source code it looks like there are some conditional functions if certain boards are defined. 
writeIndexedPixels seems to belong to only ESP32.
  
I really appreciate the work that's gone into this library the examples worked amazingly well on my ESP32; now I'm hoping to build something using the STM32411CEU.

Looking forward to your advice

Thanks

moononournation, 2021-02-01:
I have not yet tried STM32F411, did it works well on PDQgraphicstest example? Can you share the figures?

JonRobo, 2021-02-02:
I changed the board to STM32F411CEU Blackpill and kept the example unchanged.

These are the errors. 

Arduino: 1.8.12 (Windows 10), Board: "Generic STM32F4 series, BlackPill F411CE, STM32CubeProgrammer (DFU), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:82:6: error: no declaration matches 'void Arduino_DataBus::writeIndexedPixels(uint8_t*, uint16_t*, uint32_t)'
   82 | void Arduino_DataBus::writeIndexedPixels(uint8_t *data, uint16_t *idx, uint32_t len)
      |      ^~~~~~~~~~~~~~~
C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:82:6: note: no functions named 'void Arduino_DataBus::writeIndexedPixels(uint8_t*, uint16_t*, uint32_t)'
In file included from C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:5:
C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.h:60:7: note: 'class Arduino_DataBus' defined here
   60 | class Arduino_DataBus
      |       ^~~~~~~~~~~~~~~
C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:90:6: error: no declaration matches 'void Arduino_DataBus::writeIndexedPixelsDouble(uint8_t*, uint16_t*, uint32_t)'
   90 | void Arduino_DataBus::writeIndexedPixelsDouble(uint8_t *data, uint16_t *idx, uint32_t len)
      |      ^~~~~~~~~~~~~~~
C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:90:6: note: no functions named 'void Arduino_DataBus::writeIndexedPixelsDouble(uint8_t*, uint16_t*, uint32_t)'
In file included from C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.cpp:5:
C:\Users\Jonathan\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_DataBus.h:60:7: note: 'class Arduino_DataBus' defined here
   60 | class Arduino_DataBus
      |       ^~~~~~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32F4 series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
[image]

moononournation, 2021-02-02:
The above code is under this preprocessor directive:
```
#if defined(ARDUINO_ARCH_SAMD) || defined(ESP8266) || defined(ESP32)
```
and should not compiled for STM32F4...

JonRobo, 2021-02-02:
Right, I noticed this in the source code when trying the GIF example. There are many conditional statements in the source to include certain methods depending on what particular boards is used. I can try to grind through this myself, but I think it would be quite valuable to add support in this library for the STM32F4 chips. They should have enough power to play gifs like the ESP32, and I want to add a micro SD card to my project to store all Gifs/and jpgs.  The microcontroller itself can be bought for ~2 USD on Taobao, which makes it a really great option for a product BOM.

moononournation, 2021-02-02:
I have just updated the ImgViewerAnimatedGIF example, please check is it ok now.

JonRobo, 2021-02-02:
Hello!

The PDQgraphicstest example is now working, the GIF example has trouble with accessing the SD card attached on SPI bus1. 
I have yet to find a good example for using the Blackpill with an SD card which is unfortunate. Seems to be a bit more of a struggle compared to other arduino compatible boards.

Thanks for your quick action to improve the library, I'm still blown away at the quality of pictures and gifs I can run on my display (at least with the ESP32 w/DMA, having some speed issues when accessing the GIF from SD)

closed 2021-02-07 by JonRobo