moononournation/Arduino_GFX #20: Playing GIFs from SD card is Slow

JonRobo, 2021-02-07:
Using ESP32 + SD card to playback a gif. 
The playback is about 70% slower than it should be, playing from SPIFFS was much faster and exactly the same as the playback on my computer

I am noticing that the time to get the frame is slower than the allotted time determined from t_delay, fetching of the GIF frame of the SD card must be taking too long.

I have both the GC9A01 and the SD card on the VSPI bus, I'm wondering if we can get the playback fast enough or will we need to put the display and SD card on separate busses and then have two tasks running. One to fetch the next GIF frame, the other to play it back.

Thanks in advance

JonRobo, 2021-02-09:
I discovered that using the ESP32SPI_DMA bus for the display and editing the MAX_TRANSFER_SZ in Arduino_ESP32SPI_DMA.h to (32*1024) from (480 * 2 * 9) had a big impact to gif playback speed. I'm fairly satisfied with the quality now.

However now I'm noticing that high quality jpgs from the SD card are slow too haha. My test image is 67K in size and you can see the image be filled in row by row. I'll be trying to optimize this next

closed 2021-02-09 by JonRobo

moononournation, 2021-02-09:
If you are using T8 v1.8, you may try SD_MMC 4-bit. it is a little bit
faster.

JonRobo, 2021-02-09:
Using the GC9A01 and and SD breakout board. 
I noticed that for Gifs we can use writeIndexedPixels to display the whole frame right?

For Jpgs I see that we're writing row by row with draw16bitBeRGBBitmap. 

Wouldn't it be possible to grab the jpg data in such a way that we can use it in the writeIndexedPixels and basically treat it like a frame of a gif?

moononournation, 2021-02-09:
4-bit SD_MMC can play 22050 MP3 + 320x240 MJPEG over 15 FPS, is it already good enough?
https://github.com/moononournation/ESP32-LCDKit/blob/master/Mjpeg_MP3_JPEGDEC_DMA/Mjpeg_MP3_JPEGDEC_DMA.ino

JonRobo, 2021-02-10:
I found that by changing the JPG size and buff size to 4096 from 2048 in JPGDEC.h (bitbank2/JPEGDEC) library source code we get an enormous increase in speed to display jpgs. I now have only ~0.14 seconds to display a quality 240*240 jpg from an SD card. I know that these libraries are meant to run on as many microcontrollers as possible, but with the ESP32 there is an order of magnitude more RAM than most others and by utilizing it both GIF playback and image rendering can be improved substantially.
[image]

I'm now going to shift attention to custom animations such as displaying live sensor data as a graph

moononournation, 2021-02-10:
Good to see the preformance gain, I think you should told your findings to
bitbank2. He may be can change it to as a parameter for this flexibility.

moononournation, 2021-02-10:
But I said I can play 320x240 MJPEG at 15 FPS with MP3 simultaneously, or 30 FPS without MP3, that means only 0.033 seconds a frame.

JonRobo, 2021-02-10:
I'd like to test that example later on as well, does this use both cores to achieve 30 FPS? One to fill the buffer and the other core to display? 

I'll mention this change to bitbank2 as well, my findings have all been surface level; I'm sure the author would know of some additional ways to improve performance for more capable controllers like the ESP32