moononournation/Arduino_GFX #22: gfx->setRotation(0) works for jpg but not for gifs

JonRobo, 2021-02-10:
When trying to change the orientation of the display the gfx->setRotation(X); method works well for jpgs
With a strange mapping on my setup at least where 0 -> 0 degrees 1 -> 180 degrees, 2 -> 90 degrees, 3 -> 270 degrees

But when any gfx->setRotation(X) is set the gif playback is empty :/

Would appreciate any input, thank-you

moononournation, 2021-02-10:
which display driver?

JonRobo, 2021-02-10:
GC9A01, do you find the same issue?

JonRobo, 2021-02-10:
I noticed that if this is set the rotation first then display a jpg then play a gif, the gif will playback properly in the set orientation.
However if only the gif is played in the program there is nothing on the screen if I try to set the rotation

moononournation, 2021-02-10:
> I noticed that if I set the rotation first then display a jpg then play a

have you setAddrWindow after setRotation?

JonRobo, 2021-02-10:
Yes the setAddrWindow comes after the rotation
[image]

JonRobo, 2021-02-11:
Fixed this problem with a weird hack. Here is my rotation function

void set_rotation(uint16_t rotation)
{
  switch(rotation)
  {
    case 0:
      gfx->setRotation(0);
      gfx->draw16bitBeRGBBitmap(0, 0, {0x00}, 1, 1);
      break;
    case 90:
      gfx->setRotation(2);
      gfx->draw16bitBeRGBBitmap(0, 0, {0x00}, 1, 1);
      break;
    case 180:
      gfx->setRotation(1);
      gfx->draw16bitBeRGBBitmap(0, 0, {0x00}, 1, 1);
      break;
    case 270:
      gfx->setRotation(3);
      gfx->draw16bitBeRGBBitmap(0, 0, {0x00}, 1, 1);
      break;
  }
}
After changing the rotation if the display is fed even a single bit immediately and then a gif is played the gif plays in the desired orientation. In my case this pixel is blank and will show up in a non-visible part of my circular display so all is good for me now.

closed 2021-02-11 by JonRobo

moononournation, 2021-02-14:
I know what's the problem, should be fixed now.

commit 2b81d1b5 (2021-02-14): fix #22