moononournation/Arduino_GFX #23: Documentation: Arduino_HWSPI vs ESP32SPI vs ESP32SPI_DMA

marcmerlin, 2021-02-13:
First, I wanted to give a huge thumbs up for your excellent work on this library. The amount of MCU support, types of communication support and number of TFTs, is really good, thank you.

Is it possible to explain in the documentation for regular SPI (not 4bit or 8/9bit), the difference between:
- Arduino_HWSPI(TFT_DC, TFT_CS)
- Arduino_ESP32SPI(TFT_DC, TFT_CS, TFT_SCK, TFT_MOSI, TFT_MISO, VSPI /)
- Arduino_ESP32SPI_DMA(TFT_DC, TFT_CS, TFT_SCK, TFT_MOSI, TFT_MISO, VSPI);

Are they not all supposed to be using hardware SPI if I use VSPI pins? Yet, they are not all the same speed, ESP32SPI_DMA is definitely faster. Is it a better implementation of the HW SPI stack? Still, doesn't HW SPI use support inside the chip?
I'm just curious about what you made faster :)

moononournation, 2021-02-13:
The DMA driver is still in PoC status and it is not faster in many
circumstances, so not yet have document mentioned it.

marcmerlin, 2021-02-13:
Understood. That said, even Arduino_HWSPI vs Arduino_ESP32SPI isn't very clear. Is there any reason to use HWSPI or ESP32SPI is always better? If it's always better, than ESP32SPI could be default?

moononournation, 2021-02-14:
HWSPI is using Arduino built-in SPI class, so it should compatible to all Arduino hardware. ESP32SPI is using ESP32 specific implementation, so it have more options and better performance.

marcmerlin, 2021-02-14:
Right, but HWSPI is implemented by the ESP32 abstraction layer, and was fast already.
I guess you wrote your own and it's better than the one that me-no-dev wrote in arduino-esp32. Great job.
Feel free to close this issue, or leave it open until there is more info on the different options in he wiki, I'll leave it up to you. Thanks for your answer.

moononournation, 2021-02-15:
ESP32SPI is copied from arduino-esp32 SPI class. I need to implement 9-bit SPI so make some caching and the cache mechanism have a side effect of dump data faster.

closed 2021-04-24 by moononournation