moononournation/Arduino_GFX #25: please rename Data -> data

marcmerlin, 2021-02-14:
Examples like ImgViewerAnimatedGIF have a directory called Data that contains files to upload to SPIFFS or FatFS.
The expected name is 'data', not 'Data', so it doesn't work on linux unless I rename it.

It probably works on windows because windows does not care about uppercase and lowercase. On MacOS I'm not sure, I guess it works too?
That said, if you could rename as 'data', it will work on unix too :)

If you'd like me to send you a pull request, let me know, will be happy to

Thanks

moononournation, 2021-02-15:
Thx, I am also struggling which case is correct.

closed 2021-02-15 by moononournation

marcmerlin, 2021-02-15:
Thank you for the very quick fix, much appreciated. I know for sure it's 'data', not 'Data', but 'Data' can work too on some computers like yours :)
reference: https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/

commit cd730849 (2021-02-15): fix #25