moononournation/Arduino_GFX #27: ImgViewerAnimatedGIF crashes with ezgif.com-optimize.gif but works with ezgif.com-resize.gif

marcmerlin, 2021-02-14:
```
File: /ezgif.com-optimize.gif
new_table() malloc: 32776
GIF video start
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d17c1  PS      : 0x00060130  A0      : 0x800d1b55  A1      : 0x3ffb1eb0  
A2      : 0x3ffc00a4  A3      : 0x3ffb20f8  A4      : 0x40041574  A5      : 0x00000000  
A6      : 0x00000fff  A7      : 0x3ffd1cb7  A8      : 0x3ffc9d94  A9      : 0x00000000  
A10     : 0x0000028e  A11     : 0x00000093  A12     : 0x0000fefe  A13     : 0x3ffb24f8  
A14     : 0x3ffb24f8  A15     : 0x00000008  SAR     : 0x00000016  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40041578  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x400d17c1:0x3ffb1eb0 0x400d1b52:0x3ffb1f20 0x400d6ac3:0x3ffb1fb0 0x4008a529:0x3ffb1fd0

Rebooting...
```

```
PC: 0x400d17c1: GifClass::read_image_data(gd_GIF*, short, unsigned char*) at /tmp/arduino_build_303251/sketch/GifClass.h line 575
```

I changed the code to display the array element:
```
                else
		{
		    Serial.println(entry.prefix);
		    Serial.flush();
                    entry = gif->table->entries[entry.prefix];
		}
```
and I see this before it dies:
```
32
590
589
522
352
351
350
2
24149
65278
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
```

marcmerlin, 2021-02-14:
I think there are multiple issues with that gif library, I have other gifs that don't load and some that do not display properly
If you want a better gif library, I use this one (I didn't write it): https://github.com/marcmerlin/AnimatedGIFs 
There is a new version from Louis Beaudoin also: https://github.com/pixelmatix/GifDecoder/

moononournation, 2021-02-15:
Which MCU?

marcmerlin, 2021-02-15:
Oh, sorry, that was an ESP32 stack trace

moononournation, 2021-02-15:
Thx, I should know the error look like ESP32.
It is caused by data size overflow, that's why I don't like the data type "int", it should be fixed now.

commit c1fbce26 (2021-02-15): fix #27
closed 2021-02-15 by moononournation