moononournation/Arduino_GFX #38: Display Colors Inverted

3mc2, 2021-04-01:
Hello, 
Do you have a user guide available.  I need to do things like rotate the display and my colors are inverted.  I am sure they are easy corrections but I do not know the commands to do it,  Can you please help.

moononournation, 2021-04-01:
Which display are you using? You may find more parameters details at Arduino_GFX WiKi.

3mc2, 2021-04-01:
I am using a 3.5 Inch LCD with 262k colors.  It uses an ili9488 driver.  It
works with tft_espi with the inverse setting turned on.  It is working on
your but..  The colors are inverted.  Your play mjpeg is really good.  I
just need to figure out the inversion issue.

Rich

moononournation, 2021-04-02:
Added ips parameter to all Arduino_ILI948* class, set true for invert the color.

closed 2021-04-02 by moononournation