moononournation/Arduino_GFX #43: compile error using arduino IDE 1.8.13

MHz000, 2021-04-23:
got:
In member function 'virtual void Arduino_TFT::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t)':
...\libraries\Arduino_GFX-master\src\Arduino_TFT.cpp:1125:1: error: a15 cannot be used in asm here
 }
 ^
by using  https://github.com/moononournation/Arduino_GFXmoononournation one of the  Arduino_GFX_Library examples

Google find plenty issues related to that error. However I didn't have the expertise to apply the proposals.

moononournation, 2021-04-28:
I just located the error, it appear while compile for ESP8266 board at Windows platform. I will try to find out why ESP8266 compiler only have this error in Windows platform.

MHz000, 2021-04-28:
Great to localise the reason for error msg. You'll find some may be useful
hints by google.
Good luck und pls keep me informed.
Regards

moononournation, 2021-04-28:
Please get update and check is it the workaround ok for you.

MHz000, 2021-04-28:
Very fast response, thank you. However the error remains the same. See the
compiler msg here
C:\Dropbox\ARDUINO sketch
book\libraries\Arduino_GFX-master\src\Arduino_TFT.cpp: In member function
'virtual void Arduino_TFT::drawChar(int16_t, int16_t, unsigned char,
uint16_t, uint16_t)':
C:\Dropbox\ARDUINO sketch
book\libraries\Arduino_GFX-master\src\Arduino_TFT.cpp:1125:1: error: a15
cannot be used in asm here
 }
 ^
C:\Dropbox\ARDUINO sketch
book\libraries\Arduino_GFX-master\src\Arduino_TFT.cpp:1125:1: error: a15
cannot be used in asm here

All the best

moononournation, 2021-04-29:
I use another computer to test it, the problems should already been fixed in latest commit.

MHz000, 2021-04-30:
Hi moononournation
one step further but pls have a look at my last attempt
1) update to last Arduiono_GFX lib
2) set SSID & PW
3) set 1.8" GREENTAB 128x160 ~ line 154
   connect TFT Vcc, GND CS to D8
                                         RST to D3
                                         A0    to D4
                                        SDA to D7
                                        SCK to D5
4) compile & upload with success 😊
however display is not initialised (wrong wiring?)  and lots of messages
like:
[HTTPS] GET... failed, error: connection failed
adm1_new_conf: 0
adm1_cum_conf: 0
adm1_cum_death: 0
print data
updateFigures complete
last_rss_update_ms: 5628, millis(): 5728
.... and so on and on
Thak-you for support

moononournation, 2021-05-21:
4) The display may already initiated but HTTP not yet have any result to display. You may try hello world example first.

closed 2021-06-14 by moononournation