moononournation/Arduino_GFX #48: Arduino_Duplicate_Display driver support

arduino12, 2021-05-06:
**Hi,**

**Great work** @moononournation !

I want to add an `Arduino_Duplicate_Display` driver with a constructor like:
`Arduino_Duplicate_Display(Arduino_GFX *gfxs[]);`
And I want it to override all of the Arduino_GFX and Arduino_TFT needed functions so we can use it like:

```
Arduino_DataBus *bus = new Arduino_ESP32SPI(BUS_DC_PIN, -1, BUS_SCK_PIN, BUS_DAT_PIN, -1, BUS_SPI_ID);

Arduino_GFX *gc9a01 = new Arduino_GC9A01(bus, GC9A01_CS_PIN, BUS_RST_PIN); // 240x240px
Arduino_GFX *st7789 = new Arduino_ST7789(bus, ST7789_CS_PIN, BUS_RST_PIN); // 240x240px
Arduino_GFX *ssd1351 = new Arduino_SSD1351(bus, SSD1351_CS_PIN, BUS_RST_PIN); // 128x128px

Arduino_GFX *gfxs[] = {gc9a01, st7789, ssd1351, NULL};
Arduino_GFX *gfx = new Arduino_Duplicate_Display(gfxs);

void setup(void)
{
  // on all displays:
  gfx->begin();
  gfx->fillScreen(BLACK);
  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");
  // can also do:
  ssd1351->fillScreen(GREEN);
}
```

**But I have some problems:**

**1.** I need the `Arduino_TFT` and ALL the display drivers to accept the `uint8_t cs_pin` argument in their constructors.
I don't understand why the `Arduino_DataBus` needs the cs pin? I think this pin is needed per display not per bus!
**2.** Do I need to inherit from `Arduino_GFX` or `Arduino_TFT`?
The problem with `Arduino_TFT` is that it needs a bus in its constructor, but it implements more functions like:
`begin, invertDisplay, displayOn, displayOff, writeRepeat, writeAddrWindow, setAddrWindow...`
**3.** Do you think a `NULL` terminated array of displays it good for the `Arduino_Duplicate_Display` constructor?

**Looking forward,
Arad :)**

moononournation, 2021-05-06:
since most databus have data buffer, it is better each display have its own
databus instance

arduino12, 2021-05-06:
But it is more elegant that the control of the cs_pin is done by the lib not the user,
Take a look at my idea of the constructor:
```
Arduino_TFT::Arduino_TFT(
    Arduino_DataBus *bus,
    int8_t cs_pin=-1,
    int8_t bl_pin=-1,
    int8_t rst_pin=-1,
    uint8_t r, bool ips,
    int16_t w, int16_t h,
    uint8_t col_offset1, uint8_t row_offset1, uint8_t col_offset2, uint8_t row_offset2)
```
This way each display driver can control its own `cs_pin` and `bl_pin` so the user don't need to do this manually for each one.
And because of the added `bl_pin` we can add backlight control function like:
`displayBacklight(uint16_t brightness)` that will do `analogWrite` to the `bl_pin`..

If the user don't provide the `cs_pin` or the `bl_pin` - the are set to -1,
and the user can provide the `cs_pin` to the `Arduino_DataBus` so this is backward compatible :)

**Don't you agree those are good improvements?**

Because one of the main advantages of your lib is the support of multiple displays-
unlike the [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) that can only do one..

moononournation, 2021-05-06:
User don’t need concerns the CS pin operation, simply input the different
CS pin for each databus instance, gfx library will do the job. You may
refer to the MultipleDeviceTest example for more information.

closed 2021-05-21 by moononournation