moononournation/Arduino_GFX #50: “a15 cannot be used in asm here”in platform io

RichardWei, 2021-05-08:
### hello,i'm very new to use platform with arduino,  when I use this in,it show error messages,my ini files,i use the clock as the main.cpp files.
---
      [env:d1]
      platform = espressif8266
      board = d1
      framework = arduino
      lib_deps=
          moononournation/GFX Library for Arduino @ ^1.1.2
          moononournation/GFX Library for Arduino @ ~1.1.2
          moononournation/GFX Library for Arduino @ 1.1.2
--- 
error messages
---
        .pio/libdeps/d1/GFX Library for Arduino/src/Arduino_TFT.cpp:1125:1: error: a15 cannot be used in asm here
---    
will thanks for your help.

moononournation, 2021-05-10:
it should be fixed at v1.1.3

RichardWei, 2021-05-12:
Great,,it works,thanks.

closed 2021-05-13 by moononournation