moononournation/Arduino_GFX #51: Arduino_ESP32PAR8 is not allowing LCD write pin above 31 on ESP32

JoskeVdB, 2021-05-13:
Hey,

I need to use the write pin of my ILI9486 device to be above 31. This pin is shared with the touch panel and since ESP32 cannot use ADC2 together with WiFi, I have to use ADC1 (pins above 31). I have found a way to allow the use of this pin by tweaking the file "\databus\Arduino_ESP32PAR8.cpp" as follows (starting from line 331 in this file):

INLINE void Arduino_ESP32PAR8::WRITE(uint8_t d)
{
  uint32_t mask = xset_mask[d];
  *dataPortClr = dataClrMask;
  *dataPortSet = mask;
#if !defined(SET_DATA_AND_WR_AT_THE_SAME_TIME)
  *wrPortClr = wrPinMask;  //  <---- ADD THIS LINE!!!
  *wrPortSet = wrPinMask;
#endif // !defined(SET_DATA_AND_WR_AT_THE_SAME_TIME)
}

Is there any chance for you to implement this change in a future release of your code?

Thank you

moononournation, 2021-05-13:
Sorry, that is ESP32 attribute. GPIO 0-31 and 32-33 are 2 data ports.

JoskeVdB, 2021-05-13:
But by doing so, it works perfectly:

dataClrMask is only 32 bits wide, so the "dataClrMask = (1 << _wr) | ...." on line 119 gets lost when _wr is above 31.
Due to this, the wr line is not cleared (set to 0) in the command "*dataPortClr = dataClrMask;" on line 334.
By adding "*wrPortClr = wrPinMask;" just before the "*wrPortSet = wrPinMask;" allows the wr to be strobed low and high again.

moononournation, 2021-05-13:
Ok, I may consider create a separate slower PAR8 class to cater this.

JoskeVdB, 2021-05-13:
Thank you.
Indeed, adding this line will slow down things a little bit, and your gfx is very fast compared with other ones.

moononournation, 2021-05-13:
Separate class is actually straightforward, so I make a quick fix.
FYI, LovyanGFX is much faster and it does not have the WR <= 31 problems.

closed 2021-05-13 by moononournation