moononournation/Arduino_GFX #52: error: expected type-specifier before 'Arduino_NRFXSPI'

ufanders, 2021-05-19:
Hi there, I'm getting an error using the Adafruit nRF52840 Feather board:

```
error: expected type-specifier before 'Arduino_NRFXSPI'
 Arduino_DataBus *bus = new Arduino_NRFXSPI(2 /* DC */, 3 /* CS */, 14 /* SCK */, 13 /* MOSI */, 15 /* MISO */);
```

Here is my declaration:

```
//Begin TFT
#define ARDUINO_ARCH_NRF52840
#include <Arduino_GFX_Library.h>
#define TFT_BL 30 /*Backlight control, pin 'A2'.*/
Arduino_DataBus *bus = new Arduino_NRFXSPI(2 /* DC */, 3 /* CS */, 14 /* SCK */, 13 /* MOSI */, 15 /* MISO */);
Arduino_GFX *gfx = new Arduino_ST7735(bus, 28 /*RST*/, 0 /* rotation */, false /* IPS */);
//End TFT
```

The declaration appears correct for this platform, can you help? Thanks!

moononournation, 2021-05-19:
ARDUINO_ARCH_NRF52840 should be defined according to your selected board, you cannot enforce `#define ARDUINO_ARCH_NRF52840` in your sketch. I recommend you try HWSPI first instead of NRFXSPI.

ufanders, 2021-05-20:
The board I have selected is "Adafruit Feather nR52840 Express", after this post I noticed what you say and commented out the #define. =]

I will try HWSPI, thank you!

ufanders, 2021-05-20:
I solved the problem - I was using incorrect pin numbers. The correct pin-mapping is in:

`Library/Arduino15/packages/adafruit/hardware/nrf52/0.21.0/variants/feather_nrf52840_express/variant.h`

You must use a pin number that translates to the "Dx" (e.g. D1, D2, D3) Arduino convention that indexes into g_ADigitalPinMap[]. Very frustrating.

I used this and it works:

`Arduino_DataBus *bus = new Arduino_HWSPI(TFT_DC, TFT_CS, false);`

Thanks!

moononournation, 2021-05-21:
NRFXSPI is written for Arduino Nano 33 BLE, yes it run faster but not DMA.

closed 2021-05-21 by moononournation