moononournation/Arduino_GFX #53: nRf52840 Feather - No GPIO activity

ufanders, 2021-05-19:
Hi there,

I am seeing no activity on any of the GPIO I'm using to drive my ST7735S TFT. Here is my declaration:

```
//Begin TFT
#include <Arduino_GFX_Library.h>
//Override pin locations from Arduino_GFX_Library.h
//NOTE: pin numbers follow P0.xx convention, not Arduino Dx convention.
//#define TFT_SCK 14 //SCK
//#define TFT_MOSI 13 //MO
//#define TFT_MISO 15 //MI
#define TFT_CS 3 //A5
#define TFT_DC 2 //A4
#define TFT_RST 28 //A3
#define TFT_BL 30 //A2
//Arduino_DataBus *bus = new Arduino_NRFXSPI(TFT_DC, TFT_CS, TFT_SCK, TFT_MOSI, TFT_MISO);
Arduino_DataBus *bus = create_default_Arduino_DataBus();
Arduino_GFX *gfx = new Arduino_ST7735(bus, TFT_RST, 0 /* rotation */, false /* IPS */);
//End TFT
```

Here is my TFT init in setup():

```
//TFT
  gfx->begin();
  gfx->fillScreen(BLACK);

#ifdef TFT_BL
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);
#endif

  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");
```

I reviewed the pin mapping for the nRF52840 Feather board by Adafruit, and it seems correct.

Can you help me understand why the GPIO are not operating? Thanks!

moononournation, 2021-05-19:
I did not have feather board in hand, what board selected in Arduino IDE?

ufanders, 2021-05-20:
"Adafruit Feather nRF52840 Express".

ufanders, 2021-05-20:
I solved the problem - I was using incorrect pin numbers. The correct pin-mapping is in:

Library/Arduino15/packages/adafruit/hardware/nrf52/0.21.0/variants/feather_nrf52840_express/variant.h

You must use a pin number that translates to the "Dx" (e.g. D1, D2, D3) Arduino convention that indexes into g_ADigitalPinMap[]. Very frustrating.

I used this and it works:

Arduino_DataBus *bus = new Arduino_HWSPI(TFT_DC, TFT_CS, false);

Thanks!

closed 2021-05-20 by ufanders