moononournation/Arduino_GFX #56: Last row data undefined?

ufanders, 2021-05-21:
Hi there, 

I'm using a 80x160px ST7735 based TFT. My hard-coded offsets/extents are here:

```
#define TFT_OFFSETX 26
#define TFT_OFFSETY 1
#define ST7735_TFTWIDTH 80
#define ST7735_TFTHEIGHT 160
```

I draw a rectangle here:

```
gfx->fillScreen(BLACK);
gfx->drawRect(gOffsetX+TFT_OFFSETX, gOffsetY+TFT_OFFSETY, ST7735_TFTWIDTH, ST7735_TFTHEIGHT, GREEN);
```

This is the result:

[image]

I noticed there are additional offset arguments in the GFX library:

```
Arduino_ST7735(
      Arduino_DataBus *bus, int8_t rst = -1, uint8_t r = 0,
      bool ips = false, int16_t w = ST7735_TFTWIDTH, int16_t h = ST7735_TFTHEIGHT,
      uint8_t col_offset1 = 0, uint8_t row_offset1 = 0, uint8_t col_offset2 = 0, uint8_t row_offset2 = 0,
      bool bgr = true);
```

Can you help me understand why the last row contains random data when displayed?

Thanks!

moononournation, 2021-05-21:
Please find more details on wiki tab.

ufanders, 2021-05-22:
Ah, thank you so much for having many display- specific examples. It looks perfect now!

closed 2021-05-22 by ufanders