moononournation/Arduino_GFX #62: File system mount failed

hackintoshlover, 2021-06-13:
I get the following error after uploading the data and sketch to esp32
E (512) SPIFFS: mount failed, -10025
Please help in resolving the same

moononournation, 2021-06-14:
please provide more details what you are doing

hackintoshlover, 2021-06-14:
Thanks for the reply.
Basically I have interfaced gc9a01 based 1.28" round display to esp32.
I want to play video on it but I started with images.
So I opened imgviewerjpg example from the library.
Uploaded the data to eep32 SPIFFS using esp32fs plugin.
After that I compiled and upload the sketch to esp32.
But instead of showing image on the screen, it says system mount failed.
Also the same error on serial monitor.
Please help.
Also I want to play a video file on it, please guide me through the same.

moononournation, 2021-06-14:
try upload program again after flash SPIFFS

hackintoshlover, 2021-06-14:
I did try that but it still did not work

hackintoshlover, 2021-06-14:
What should I do to resolve the issue?

hackintoshlover, 2021-06-14:
Alternately is there a way to connect a memory card and play video from it on gc9a01 using esp32?

moononournation, 2021-06-14:
check the code is using FatFS or SPIFFS

hackintoshlover, 2021-06-14:
How do I check that?
Also I am using the default code provided in the example folder of your library

hackintoshlover, 2021-06-14:
Just an update
In void setup()
I changed line 98 from if (!SPIFFS.begin()) to if (!SPIFFS.begin(true))
and the code uploads fine and i get no SPIFFS error, but nothing displays on the screen
Please help

moononournation, 2021-06-14:
any error when upload SPIFFS

hackintoshlover, 2021-06-14:
No error now
But the screen just stays blank and displays nothing
Also tried adding a 240*240px image
But still blank screen
Any help on this would've really helpful

moononournation, 2021-06-14:
it should have some error while you upload the SPIFFS file data

hackintoshlover, 2021-06-14:
Attaching screenshots
[image]
[image]

No compile error or upload error
Also attached serial monitor output

hackintoshlover, 2021-06-14:
Please help to resolve

moononournation, 2021-06-14:
how about "ESP32 Sketch Data Upload"

hackintoshlover, 2021-06-14:
> How about the message of "ESP32 Sketch Data Upload"

It say "uploaded successfully"

hackintoshlover, 2021-06-14:
> I also find the latest arduino-esp32 have this problem if selected default partition scheme. Try select another partition scheme, e.g. No OTA (2MB APP/2MB SPIFFS).

Tried this but no luck

moononournation, 2021-06-14:
you require upload program and data again after change partition scheme

hackintoshlover, 2021-06-14:
did it already
still same issue

hackintoshlover, 2021-06-14:
Is there any way of uploading data to spiffs memory externally?
without using arduino ide
via commandline?

hackintoshlover, 2021-06-14:
Can you tell me which arduino version and esp32 arduino core version you are using so that i can try the same

hackintoshlover, 2021-06-14:
Hey, the error got fixed after i updated esp32 arduino core version to 1.0.6

closed 2021-06-14 by hackintoshlover

moononournation, 2021-06-14:
If you trying latest version, I recommend you change to use FFat, it is faster in many circumstances.