moononournation/Arduino_GFX #64: Invalid conversion

hackintoshlover, 2021-06-14:
Following your instructables - https://www.instructables.com/Play-Video-With-ESP32/
Wired everything up.

When i compile the code with default settings no errors are encountered and arduino says done compiling.
But when i change the screen defines to GC9A01 i get error saying invalid conversion.

See attached images below for clarity.
Please help me with this since i have a GC9A01 based round display.

I am using arduino ide 1.8.13 and esp arduino core version 1.0.6

[image]
[image]
[image]
[image]

moononournation, 2021-06-14:
replace Arduino_GFX *gfx to Arduino_TFT *gfx

hackintoshlover, 2021-06-15:
Thanks for the solution, it compiled perfectly
I have a small problem though
Wired everything as per Following your instructables - https://www.instructables.com/Play-Video-With-ESP32/
I get sd card not found
i am using 
https://circuitstate.com/tutorials/interfacing-catalex-micro-sd-card-module-with-arduino/
memory card module
schematics for the same https://circuitstate.com/wp-content/uploads/2021/05/Catalex-Micro-SD-Card-Module-Schematic.png

The card reader module is working fine when used with builtin sdcard codes

hackintoshlover, 2021-06-15:
I think so cs pin needs to be connected somewhere

hackintoshlover, 2021-06-15:
sd card mount failed!
[image]

moononournation, 2021-06-15:
if you want to try the SD card module, select the examples with prefix SD_*

hackintoshlover, 2021-06-15:
Tried all the examples you have provided
Constantly getting error sdcard mount failed
Please help me with this

hackintoshlover, 2021-06-15:
I am using micro SD card and wired it in 1 bit more as per suggested by you in the instructables

moononournation, 2021-06-15:
it should works: https://youtu.be/mVriPOfqwtk

hackintoshlover, 2021-06-15:
But it is not working
can you suggest something regarding the same
Also i am using micro sdcard
Should i use big sd card

moononournation, 2021-06-15:
you may try another card, I know some random card doesn't work

hackintoshlover, 2021-06-15:
Tried 3 cards
Also when i use this card with arduino uno it works

moononournation, 2021-06-15:
I do not use a dedicated SD card module, I have no idea on it.

hackintoshlover, 2021-06-15:
ok ill also check with just sd card and let you know

hackintoshlover, 2021-06-15:
Finally got it to work
pinMode(2,INPUT_PULLUP);
pinMode(4,INPUT_PULLUP);
pinMode(12,INPUT_PULLUP);
pinMode(13,INPUT_PULLUP);
pinMode(14,INPUT_PULLUP);
pinMode(15,INPUT_PULLUP);
And added 1k resistor between 2 and 3.3v
But the video output is not as smooth as yours
attaching video for your reference
please help me in getting a smooth output, suggest what steps to take
Attached link contains the video output from esp32
https://drive.google.com/file/d/1X81rjiPrlJhKMVROIcJCQM8CVzNZ_GpW/view?usp=drivesdk

hackintoshlover, 2021-06-16:
Hey please help

hackintoshlover, 2021-06-16:
Can you share the video that you used to try and see if that works in my case

moononournation, 2021-06-16:
Please follow the ffmpeg script in README.md to convert your video: https://github.com/moononournation/RGB565_video

hackintoshlover, 2021-06-16:
i have already followed the ffmeg script to convert the video, still facing the issue
I tried everything mjpeg, rgb565, etc
Still same output
What should i do next
Please help
Also is there a way to use the screen with raspberry pi zero by any chance

moononournation, 2021-06-16:
what is your video output resolution?

hackintoshlover, 2021-06-17:
240*240 pixels
As per supported by GC9A01 round lcd I am using currently

moononournation, 2021-06-17:
ok, then try to reduce the fps, say 5 fps

closed 2021-07-05 by moononournation