moononournation/Arduino_GFX #65: Not an issue..

nodoubtman, 2021-06-18:
Hello, not an issue here.

Is your library supports this model: ILI9486 PARALLEL 8 bits display?

And how to use it?
here's the picture included.

Thanks.
M
[image]
arc.

moononournation, 2021-06-18:
I have no this module. Should able to support this but requires create a new Arduino_ILI9486.cpp class.

closed 2021-07-05 by moononournation

moononournation, 2022-10-11:
Arduino_ILI9486 is now available, you may test it with init code something like this:
```
Arduino_DataBus *bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, GFX_NOT_DEFINED /* RD */,
    16 /* D0 */, 17 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);
Arduino_GFX *gfx = new Arduino_ILI9486(bus, 33 /* RST */, 0 /* rotation */, false /* IPS */);
```