moononournation/Arduino_GFX #68: ARDUINO MEGA2560 Not Working! :(

nodoubtman, 2021-07-18:
Hello, here's my definitions with ARDUINO MEGA2560 board:

#define TFT_RST      8
#define TFT_DC       9
#define TFT_CS      53  // SS
#define TFT_SDI     51  // MOSI
#define TFT_CLK     52  // SCK
#define TFT_LED      0  // 0 if wired to +5V directly

Arduino_DataBus *bus = new Arduino_HWSPI(TFT_DC, TFT_CS);
//Arduino_DataBus *bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_CLK, TFT_SDI, -1);

// ILI9225 LCD 176x220
Arduino_ILI9225 *gfx = new Arduino_ILI9225(bus, TFT_RST);

it seems that it doesnt work..

can you help me?
Thanks.
Marc.

moononournation, 2021-07-18:
I found you changed SWSPI patameters, do you actually want SWSPI? then you should comment HWSPI and uncomment SWSPI.

nodoubtman, 2021-07-18:
I would like to use SWSPI like that:

#define TFT_CS      10  // SS
#define TFT_SDI     11  // MOSI
#define TFT_CLK     13  // SCK
#define TFT_RST      8
#define TFT_DC       9
Arduino_DataBus *bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_CLK, TFT_SDI, -1);
// ILI9225 LCD 176x220
Arduino_ILI9225 *gfx = new Arduino_ILI9225(bus, TFT_RST);

but it doesnt but it is working with this library:  https://github.com/Nkawu/TFT_22_ILI9225/wiki

Thanks.
Marc.

nodoubtman, 2021-07-18:
Seems to work now with Arduino_SWSPI :)

Thanks for your time :)
Marc.

closed 2021-07-18 by nodoubtman