moononournation/Arduino_GFX #78: About LVGL use

modi12jin, 2021-09-05:
>  Use these libraries

```
https://github.com/moononournation/Arduino_GFX

https://github.com/DustinWatts/FT6236

https://github.com/lvgl/lvgl

https://github.com/lvgl/lv_demos
```

[image]

```
#include <lvgl.h>
#include <lv_demo.h>
#include <Arduino_GFX_Library.h>
#include <FT6236.h>
#include <Arduino.h>

#define TFT_BL 23
Arduino_ESP32SPI *bus = new Arduino_ESP32SPI(21 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */,-1);
Arduino_GFX *gfx= new Arduino_ST7796(bus, 22 /* RST */, 3 /* rotation */, false /* IPS */);

FT6236 ts = FT6236();

/*更改为您的屏幕分辨率*/
static const uint32_t screenWidth  = 480;
static const uint32_t screenHeight = 320;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[ screenWidth * 10 ];

#if LV_USE_LOG != 0
/* Serial debugging */
void my_print( lv_log_level_t level, const char * file, uint32_t line, const char * fn_name, const char * dsc )
{
   Serial.printf( "%s(%s)@%d->%s\r\n", file, fn_name, line, dsc );
   Serial.flush();
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p) {
  int w = (area->x2 - area->x1 + 1);
  int h = (area->y2 - area->y1 + 1);

  gfx->startWrite(); /* Start new TFT transaction */
  gfx->setAddrWindow(area->x1, area->y1, w, h); /* set the working window */

  gfx->writePixels(( uint16_t * )&color_p->full, w*h);

  gfx->endWrite(); /* terminate TFT transaction */
  lv_disp_flush_ready(disp); /* tell lvgl that flushing is done */
}

/*Read the touchpad*/
void my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data) {
   uint16_t touchX, touchY;

   TS_Point p = ts.getPoint();
   touchX=screenWidth-p.y;
   touchY=p.x;

   if( !ts.touched() )
   {
      data->state = LV_INDEV_STATE_REL;
   }
   else
   {
      data->state = LV_INDEV_STATE_PR;

      /*Set the coordinates*/
      data->point.x = touchX;
      data->point.y = touchY;

      Serial.print( "Data x " );
      Serial.println( touchX );

      Serial.print( "Data y " );
      Serial.println( touchY );
   }

}

void setup()
{
   Serial.begin( 115200 ); /* prepare for possible serial debug */
   Serial.println( "Hello Arduino! (V8.0.X)" );
   Serial.println( "I am LVGL_Arduino" );

   lv_init();

#if LV_USE_LOG != 0
   lv_log_register_print_cb( my_print ); /* register print function for debugging */
#endif

   gfx->begin();/* TFT init */
   gfx->setRotation( 3 ); /* Landscape orientation, flipped */

   pinMode(TFT_BL, OUTPUT);
   digitalWrite(TFT_BL, HIGH);

   //begin(40, SDA, SCL) for custom pins
   if (!ts.begin(40,18,19)){
        Serial.println("Unable to start the capacitive touchscreen.");
    }

   lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * 10 );

   /*Initialize the display*/
   static lv_disp_drv_t disp_drv;
   lv_disp_drv_init( &disp_drv );
   /*Change the following line to your display resolution*/
   disp_drv.hor_res = screenWidth;
   disp_drv.ver_res = screenHeight;
   disp_drv.flush_cb = my_disp_flush;
   disp_drv.draw_buf = &draw_buf;
   lv_disp_drv_register( &disp_drv );

   /*Initialize the (dummy) input device driver*/
   static lv_indev_drv_t indev_drv;
   lv_indev_drv_init( &indev_drv );
   indev_drv.type = LV_INDEV_TYPE_POINTER;
   indev_drv.read_cb = my_touchpad_read;
   lv_indev_drv_register( &indev_drv );

#if 0
   /* Create simple label */
   lv_obj_t *label = lv_label_create( lv_scr_act() );
   lv_label_set_text( label, "Hello Arduino! (V8.0.X)" );
   lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
#else
   /* Try an example from the lv_examples Arduino library
      make sure to include it as written above.
   lv_example_btn_1();
   */

   // uncomment one of these demos
   lv_demo_widgets();            // OK
   // lv_demo_benchmark();          // OK
   // lv_demo_keypad_encoder();     // works, but I haven't an encoder
   // lv_demo_music();              // NOK
   // lv_demo_printer();
   // lv_demo_stress();             // seems to be OK
#endif
   Serial.println( "Setup done" );
}

void loop()
{
   lv_timer_handler(); /* let the GUI do its work */
   delay( 5 );
}
```

@moononournation  I am a newbie, how to solve the error

moononournation, 2021-09-08:
I will try to write a example using lvl

modi12jin, 2021-09-09:
@moononournation  Looking forward to your LVGL instance

I have successfully ported to this
 https://github.com/Links2004/arduinoVNC/issues/23

modi12jin, 2021-09-13:
[image]

@moononournation I saw on Twitter that you succeeded. Can you provide some sample code?

moononournation, 2021-09-13:
Yes but not yet. The lvgl Arduino example is out of maintenance, I need many hack to compile it.

moononournation, 2021-10-01:
added LVGL hello world example, more to come.

closed 2021-10-01 by modi12jin