moononournation/Arduino_GFX #79: Can ESP32-S2 support 8-bit parallel port?

modi12jin, 2021-09-07:
ESP32-S2 can be added to support 8-bit parallel port

moononournation, 2021-09-08:
Yes, it is similar to RPi Pico. I just not yet have time connect test it. Can you help this?

moononournation, 2021-10-02:
I have just added a new Arduino_ESP32S2PAR8 data bus, would you please help to check is it work for you?

modi12jin, 2021-10-02:
@moononournation success

https://user-images.githubusercontent.com/40233017/135723759-daa3d281-d980-4d99-b291-26cbf427db6c.mp4

```
#include <Arduino_GFX_Library.h>

#define TFT_CS -1
#define TFT_DC 41
#define TFT_RST -1
#define TFT_RD -1
#define TFT_WR 40 
#define TFT_BL -1

// ESP32S2 parallel 8-bit
// Display D0-D15 connect to GPIO 0-7
Arduino_DataBus *bus = new Arduino_ESP32S2PAR8(TFT_DC, TFT_CS, TFT_WR /* WR */, TFT_RD /* RD */);

Arduino_GFX *gfx = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, true /* IPS */);

/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    gfx->begin();
    gfx->fillScreen(BLACK);

#ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
#endif

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(5000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    delay(1000); // 1 second
}
```

closed 2021-10-02 by modi12jin