moononournation/Arduino_GFX #80: Compile error (with strict compiler settings)

dl9rdz, 2021-09-13:
When compiling the most recent version with platformio (which by default uses strict compiler settings that fail with error upon some compiler warnings), this error arises:

Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8347C.cpp.o
In file included from .pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.cpp:5:0:
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h: In constructor 'Arduino_HWSPI::Arduino_HWSPI(int8_t, int8_t, int8_t, int8_t, int8_t, SPIClass*, bool)':
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:60:23: error: 'Arduino_HWSPI::_miso' will be initialized after [-Werror=reorder]
int8_t _sck, _mosi, _miso;
^
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:56:13: error: 'SPIClass* Arduino_HWSPI::_spi' [-Werror=reorder]
SPIClass _spi;
^
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.cpp:27:1: error: when initialized here [-Werror=reorder]
Arduino_HWSPI::Arduino_HWSPI(int8_t dc, int8_t cs / = -1 /, int8_t sck / = -1 /, int8_t mosi / = -1 /, int8_t miso / = -1 */, SPIClass spi, bool is_shared_interface / = true */)
^
Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8347D.cpp.o

moononournation, 2021-09-16:
do you encounter this error in Arduino IDE

dl9rdz, 2021-09-16:
yes

moononournation, 2021-09-18:
merged

closed 2021-09-18 by moononournation