moononournation/Arduino_GFX #84: M5stack core with specific device declaration as specified fails.

linxcow, 2021-09-19:
`#include <Arduino_GFX_Library.h>

#include <Arduino_GFX_Library.h>
// Settings Below are for the M5Stack CORE version 
// see https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration
#define TFT_BL 32
Arduino_DataBus *bus = new Arduino_ESP32SPI(27 /* DC */, 14 /* CS */, SCK, MOSI, MISO);
Arduino_GFX *gfx = new Arduino_ILI9341_M5STACK(bus, 33 /* RST */, 1 /* rotation */);

/* more fonts at: https://github.com/moononournation/ArduinoFreeFontFile.git */
#include "FreeMono8pt7b.h"
#include "FreeSansBold10pt7b.h" 
#include "FreeSerifBoldItalic12pt7b.h"
#include "FreeMono36pt7b.h"`

ERROR :

`HelloWorldGfxfont:22:24: error: expected type-specifier before 'Arduino_ILI9341_M5STACK'
 Arduino_GFX *gfx = new Arduino_ILI9341_M5STACK(bus, 33 /* RST */, 1 /* rotation */);
                        ^
exit status 1
expected type-specifier before 'Arduino_ILI9341_M5STACK'
`
Using latest Arduino GFX library 1.1.5 Arduino IDE 1.8.15

What could i do to fix this?

linxcow, 2021-09-19:
Found the fix : 

CHANGE THIS :  `Arduino_GFX *gfx = new Arduino_ILI9341_M5STACK(bus, 33 , 1);`
TO :  `Arduino_GFX *gfx = new Arduino_ILI9342(bus, 33 /* RST */, 0 /* rotation */);`

Following link needs to be updated with correct information for M5STACK: 

https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration

moononournation, 2021-09-20:
Thx, fixed.

closed 2021-09-20 by moononournation