moononournation/Arduino_GFX #88: Pi pico explorer display

omikron88, 2021-10-02:
Hello,
How to make Arduino_GFX work with Pi pico Explorer?

https://shop.pimoroni.com/products/pico-explorer-base

It uses the display that is supported I think, main problem is, that it uses MISO pin as Data/Command for display.

https://cdn.shopify.com/s/files/1/0174/1800/files/pico_explorer_schematic.pdf?v=1619077703

Is there a workaround for this?

Tom

moononournation, 2021-10-03:
I think you may follow their source use PIN_UNUSED:
```
static const uint8_t PIN_UNUSED   = UINT8_MAX;
Arduino_DataBus *bus = new Arduino_RPiPicoSPI(27 /* DC */, 17 /* CS */, PIN_SPI0_SCK /* SCK */, PIN_SPI0_MOSI /* MOSI */, PIN_UNUSED /* MISO */, spi0 /* spi */);
```

omikron88, 2021-10-10:
I had to create new derived class, with changed SPI initialisation. Because miso pin is used as dc and not miso.

void Explorer_RPiPicoSPI::begin(int32_t speed /* = 0 */, int8_t dataMode /* = -1 */)
{
  // set SPI parameters
  _speed = speed ? speed : SPI_DEFAULT_FREQ;
  _dataMode = (dataMode >= 0) ? dataMode : SPI_MODE0;
  _spis = SPISettings(_speed, _bitOrder, _dataMode);

  // set pin mode
  pinMode(_dc, OUTPUT);
  digitalWrite(_dc, HIGH); // Data mode
  if (_cs >= 0)
  {
    pinMode(_cs, OUTPUT);
    digitalWrite(_cs, HIGH); // disable chip select
  }

  if (_miso >= 0)
  {
    gpio_set_function(_miso, GPIO_FUNC_SPI);
  }
  
  gpio_set_function(_sck, GPIO_FUNC_SPI);
  gpio_set_function(_mosi, GPIO_FUNC_SPI);

  // set fastIO variables
  _dcPinMask = digitalPinToBitMask(_dc);
  _dcPortSet = (PORTreg_t)&sio_hw->gpio_set;
  _dcPortClr = (PORTreg_t)&sio_hw->gpio_clr;

  if (_cs >= 0)
  {
    _csPinMask = digitalPinToBitMask(_cs);
    _csPortSet = (PORTreg_t)&sio_hw->gpio_set;
    _csPortClr = (PORTreg_t)&sio_hw->gpio_clr;
  }

  spi_init(_spi, _spis.getClockFreq());
  spi_set_format(_spi, 8, SPI_CPOL_0, SPI_CPHA_0, SPI_MSB_FIRST);
}

Now the display blinks a little at initialisation, but it still show nothing.
Is it possible, that the display needs different init commands?

The init calls are:

Arduino_DataBus *bus = new Explorer_RPiPicoSPI(PIN_SPI0_MISO /* dc */, PIN_SPI0_SS /* cs */, PIN_SPI0_SCK /* sck */, 
    PIN_SPI0_MOSI /* mosi */, -1 /* miso */, spi0 /* spi */);
Arduino_GFX *gfx = new Arduino_ST7789(bus, -1 /* RST */, 2 /* rotation */, true /* IPS */,
    240 /* width */, 240 /* height */, 0 /* col offset 1 */, 80 /* row offset 1 */);

//------------------------------------------------------------------------------
void setup() {
  Serial.begin(9600);
  while (!Serial) { ; }

//  Wire.setSDA(SDA);
//  Wire.setSCL(SCL);
//  Wire.setClock(40000);
//  Wire.begin();

//  SPI.setRX(MISO);
//  SPI.setTX(MOSI);
//  SPI.setSCK(SCK);
//  SPI.setCS(SS);
//  SPI.begin();

  gfx->begin();
  gfx->fillScreen(BLACK);
  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");

  delay(5000);
}

moononournation, 2021-10-10:
I have just tested your case, you just need this init code:
```
Arduino_DataBus *bus = new Arduino_RPiPicoSPI(16 /* DC */, 17 /* CS */, 18 /* SCK */, 19 /* MOSI */, UINT8_MAX /* MISO */, spi0 /* spi */);
```

[image]

moononournation, 2021-10-14:
I have a Waveshare Pico LCD 1.3 and have similar declaration:
```
#include <Arduino_GFX_Library.h>
#define TFT_BL 13
Arduino_DataBus *bus = new Arduino_RPiPicoSPI(8 /* DC */, 9 /* CS */, 10 /* SCK */, 11 /* MOSI */, UINT8_MAX /* MISO */, spi1 /* spi */);
*/);
Arduino_GFX *gfx = new Arduino_ST7789(bus, 12 /* RST */, 3 /* rotation */, true /* IPS */, 240, 240, 0, 80);
```
If your test is positive, I will update the WiKi about it:
https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration

moononournation, 2022-02-09:
@omikron88, any update?

closed 2022-02-18 by moononournation