moononournation/Arduino_GFX #89: Can't compile for Raspberry Pi Pico or Arduino Nano RP2040

w5teen, 2021-10-04:
Selecting Arduino_Databus as Arduino_RPiPicoSPI() or Arduino_SWSPI() I am seeing the following errors
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:67: error: 'PIN_SPI0_SCK' was not declared in this scope
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                   ^~~~~~~~~~~~
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:67: note: suggested alternative: 'PIN_SPI_SCK'
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                   ^~~~~~~~~~~~
                                                                   PIN_SPI_SCK
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:95: error: 'PIN_SPI0_MOSI' was not declared in this scope
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                                               ^~~~~~~~~~~~~
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:95: note: suggested alternative: 'PIN_SPI_MOSI'
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                                               ^~~~~~~~~~~~~
                                                                                               PIN_SPI_MOSI
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:124: error: 'PIN_SPI0_MISO' was not declared in this scope
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                                                                            ^~~~~~~~~~~~~
/Users/wjsteen/Documents/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_RPiPicoSPI.h:16:124: note: suggested alternative: 'PIN_SPI_MISO'
   Arduino_RPiPicoSPI(int8_t dc = -1, int8_t cs = -1, int8_t sck = PIN_SPI0_SCK, int8_t mosi = PIN_SPI0_MOSI, int8_t miso = PIN_SPI0_MISO, spi_inst_t *spi = spi0); // Constructor
                                                                                                                            ^~~~~~~~~~~~~
                                                                                                                            PIN_SPI_MISO
I am not clear from your instructions where PIN_SPI0_SCK, PIN_SPI0_MOSI, PIN_SPI0_MISO are meant to be declared.

I am also seeing many warnings for unused variables although these do not appear to stop the Clock example working on a Nano 33 BLE.

moononournation, 2021-10-05:
I am using this repository: https://github.com/earlephilhower/arduino-pico.git

w5teen, 2021-10-05:
Thank you - that works. I was using the Arduino Mbed OS RP2040 Boards>Raspberry Pi Pico!

closed 2021-10-05 by w5teen