moononournation/Arduino_GFX #93: No Screen Writes after Sd.Begin()

KCNilssen, 2021-11-09:
There seems to be an issue with SD.begin();
I have tried setting the sd cs high and the display cs low, and that didnt work. 

No matter what I do, the only way to get anything to display on the screen after a call to SD.begin is to eliminate the SD.begin call all together, which is a problem since I can no longer use my sd card.

I am using an ESP32-s2.

below is a simplified sketch of my problem.

```
#include <SPI.h>                  // This library allows communication with SPI devices
#include <SD.h>                   
#include <WiFi.h>
#include <Arduino_GFX_Library.h>

#define TFT_CS 1 // -1 for display without CS pin
#define TFT_DC 3
#define TFT_RST -1
#define TFT_BL -1

Arduino_DataBus *bus = new Arduino_ESP32SPI(TFT_DC, TFT_CS, 36 /* SCK */, 35 /* MOSI */, 37 /* MISO */, HSPI /* spi_num */);

Arduino_GFX *gfx = new Arduino_ILI9341(bus, TFT_RST, 1 /* rotation */, false /* IPS */);

const int chipSelect = 33;

void setup() {
  pinMode(37,INPUT_PULLUP);
  bool setupError = false;
	Serial.begin(115200);
  delay(2000);

  // adc_power_off();
  WiFi.disconnect(true);  // Disconnect from the network
  WiFi.mode(WIFI_OFF);    // Switch WiFi off

  gfx->begin();

  gfx->fillScreen(BLACK);
  gfx->setCursor(235,225);
	gfx->print(F("Beta "));
	gfx->print(F("v "));
  gfx->println(versionNumber);

  gfx->setCursor(0, 100);     // Start at top-left corner
  // gfx->setTextSize(1);

  delay(500);

  Serial.println(F("Before sd card setup"));

  // This will display on the tft
  gfx->print(F("This does show up"));

  /************ Sd Card Setup ***************/
  gfx->setCursor(175, gfx->getCursorY() +2);

  if (!SD.begin(chipSelect)) {
    // Serial statement works fine but no text is displayed on TFT from here
    Serial.println(F("Card init. failed!"));
		gfx->setTextColor(RED);
		gfx->println(F("Failed"));
    gfx->setTextColor(WHITE);
  } else {
    // Serial statement works fine but no text is displayed on TFT from here
		Serial.println(F("Card init. Success"));
		gfx->setTextColor(GREEN);
		gfx->print(F("Success"));
		gfx->setTextColor(WHITE);
  }

  // No text is displayed on TFT from this as well
  gfx->print(F("This does not show up"));

}
```

moononournation, 2021-11-09:
I have not tried ESP32-S2 with SD, but I think it should similar to this example:
https://github.com/moononournation/RGB565_video/blob/master/SD_RGB565_video/SD_RGB565_video.ino

KCNilssen, 2021-11-09:
@moononournation Thank you for the quick reply! The only two differences I can see are line
```14  Arduino_HWSPI *bus = new Arduino_HWSPI(27 /* DC */, 14 /* CS */, SCK, MOSI, MISO); ```

and line

```55  if (!SD.begin(SS, SPI, 80000000)) /* SPI bus mode */ ```

Upon trying multiple combinations It seems that the first one (line 14, 18, 23, 35) with using HWSPI instead of Arduino_ESP32SPI in my program, line 11, does the trick.

``` Arduino_DataBus *bus = new Arduino_ESP32SPI(TFT_DC, TFT_CS, 36 /* SCK */, 35 /* MOSI */, 37 /* MISO */, HSPI /* spi_num */); ```

This change has made it work. The updated code below works just as intended! Thank you for the help @moononournation  :)

```
#include <SPI.h>                  // This library allows communication with SPI devices
#include <SD.h>                   
#include <WiFi.h>
#include <Arduino_GFX_Library.h>

#define TFT_CS 1 // -1 for display without CS pin
#define TFT_DC 3
#define TFT_RST -1
#define TFT_BL -1

Arduino_HWSPI *bus = new Arduino_HWSPI(TFT_DC /* DC */, TFT_CS /* CS */, 36, 35, 37);

Arduino_GFX *gfx = new Arduino_ILI9341(bus, TFT_RST, 1 /* rotation */, false /* IPS */);

const int chipSelect = 33;

void setup() {
  pinMode(37,INPUT_PULLUP);
  bool setupError = false;
  Serial.begin(115200);
  delay(2000);

  // adc_power_off();
  WiFi.disconnect(true);  // Disconnect from the network
  WiFi.mode(WIFI_OFF);    // Switch WiFi off

  gfx->begin();

  gfx->fillScreen(BLACK);
  gfx->setCursor(235,225);
  gfx->print(F("Beta "));
  gfx->print(F("v "));
  gfx->println(versionNumber);

  gfx->setCursor(0, 100);     // Start at top-left corner
  // gfx->setTextSize(1);

  delay(500);

  Serial.println(F("Before sd card setup"));

  // This will display on the tft
  gfx->print(F("This does show up"));

  /************ Sd Card Setup ***************/
  gfx->setCursor(175, gfx->getCursorY() +2);

  if (!SD.begin(chipSelect)) {
    // Serial statement works and text is now also displayed on TFT
    Serial.println(F("Card init. failed!"));
    gfx->setTextColor(RED);
    gfx->println(F("Failed"));
    gfx->setTextColor(WHITE);
  } else {
    // Serial statement works and text is now also displayed on TFT
    Serial.println(F("Card init. Success"));
    gfx->setTextColor(GREEN);
    gfx->print(F("Success"));
    gfx->setTextColor(WHITE);
  }

  // This now shows up!
  gfx->print(F("This now show up"));

}
```

closed 2021-11-09 by KCNilssen

frankcohen, 2022-05-20:
Thank you both for the above solution. It works successfully on a Adafruit Feather ESP32-S3-Mini and TFT 240x240 Arduino_GC9A01 display! Thanks! -Frank