moononournation/Arduino_GFX #98: Error: expected type-specifier before 'Arduino_ESP32PAR8' using ESP32 with R61529 chip

drewgandy, 2021-12-03:
Hi, I am trying to use the basic Hello World example with an ESP32 and a 8-bit parallel display using the R61529 chip.  

I have the following declarations:
```
Arduino_DataBus* bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, -1 /* RD */, 16 /* D0 */, 17 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);
Arduino_GFX* gfx = new Arduino_R61529(bus, 32, 0 /* rotation */, true /* IPS */);
```

and am getting this error:

> eSPI_Test2.ino: 27:28: error: expected type-specifier before 'Arduino_ESP32PAR8
>    Arduino_DataBus* bus = new Arduino_ESP32PAR8(27 \* DC *\, 5 \* CS *\, 26 \* WR *\, -1 \* RD *\, 16 \* D0 *\, 17 \* D1 *\, 18 \* D2 *\, 19 \* D3 *\, 21 \* D4 *\, 12 \* D5 *\, 23 \* D6 *\, 25 \* D7 *\)

Do you have any insight into what is causing this issue?

I am using Visual Micro in Visual Studio, but the same error appears in the Arduino IDE.

[This is the display](https://www.aliexpress.com/item/32940608152.html?spm=a2g0o.productlist.0.0.66e773f6vAhLCI&algo_pvid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a&algo_expid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a-16&btsid=25887317-95e8-4f7a-aabc-1913fdfeaadf&ws_ab_test=searchweb0_0,searchweb201602_6,searchweb201603_53)

Thanks,

moononournation, 2021-12-03:
what dev board you selected?

drewgandy, 2021-12-03:
[The ESP32 Dev Board ](https://www.amazon.com/gp/product/B07Q576VWZ/ref=ask_ql_qh_dp_hza).  Issue happens whether I select the Node32s or ESP32 Dev Module in the IDE.

moononournation, 2021-12-03:
just checked with latest code no this compilation error. You may try get the latest Arduino_GFX and double check no duplicate folders under Arduino/libraries

drewgandy, 2021-12-03:
Turns out I had some old cpp files in subfolders that were causing conflicts.  Once those were deleted it compiled fine.  Sorry for the confusion and I appreciate your responsiveness.

The 'Hello World' still isn't showing up on the screen (even though @daumemo's [bit-bang test program](https://github.com/daumemo/ESP32_LCD_MIPI_DBI_TYPE_B_TEST/blob/master/ESP32-specific.cpp) worked) so I will try some different pin configurations.

drewgandy, 2021-12-04:
As a follow up, I loaded up the Hello World example, using the default bus pins in your wiki, with the only change being the reset pin in the GFX parameters, using pin 15 instead of pin 7 because pin 7 is part of the onboard SPI flash.

```
Arduino_DataBus *bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, -1 /* RD */,
    16 /* D0 */, 17 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);
Arduino_GFX *gfx = new Arduino_R61529(bus, 15 /* RST */, 0 /* rotation */, true /* IPS */);
```

The program uploads fine, but the screen is totally blank (backlight does come on) and 'HELLO WORLD' does not show up.

Any thoughts on what the issue may be?

moononournation, 2021-12-04:
I remember all unused pins require pullup or pulldown according to the datasheet.

drewgandy, 2021-12-04:
Interesting. I'll take a look.  I didn't catch that when looking at daumemo's blog.  Thanks again!

ericcastro, 2021-12-20:
I am also having this problem but I checked for duplicate folders and everything seems fine. I can't get rid of it. any ideas ?

moononournation, 2022-02-09:
If your dev board have PSRAM, you should avoid those pins, e.g.:
```
Arduino_DataBus *bus = new Arduino_ESP32PAR8(TFT_DC, TFT_CS, 25 /* WR */, 32 /* RD */, 23 /* D0 */, 19 /* D1 */, 18 /* D2 */, 26 /* D3 */, 21 /* D4 */, 4 /* D5 */, 0 /* D6 */, 2 /* D7 */);
```

ericcastro, 2022-02-09:
to update on this issue, I was able to make it work in a different computer where my arduino setup is alot simpler.

this makes me believe that problem arises from conflicts with other libraries, although I couldn't say which one for now.

moononournation, 2022-02-09:
glad to hear that

closed 2022-02-09 by moononournation