moononournation/Arduino_GFX #101: Unable to use SD Card and Display together with Arduino Nano

johnvanlla, 2021-12-07:
Hi,

I am trying to work with the Adafruit SD Card breakout board, and the 1.28" Round display using the examples provided by the library. In particular the imgvieweranimatedGif example. Which pins are the SD Card CS pin suppose to be connected to? I have been able to confirm that they work independent of one another but am unable to have them work in unison. Any help would be greatly appreciated!

moononournation, 2021-12-09:
this code demonstrates how to share SPI between SD and display:

https://github.com/moononournation/RGB565_video/blob/master/SD_GIF_video/SD_GIF_video.ino

moononournation, 2021-12-12:
Sorry, I just remember Arduino Nano not have enough resources to play GIF. You may try the ImgViewerBmp.ino instead:
https://github.com/moononournation/Arduino_GFX/blob/master/examples/ImgViewer/ImgViewerBmp/ImgViewerBmp.ino

moononournation, 2022-02-09:
@johnvanlla, any update?