moononournation/Arduino_GFX #102: Cannot get Pi Pico to work with Parallel 8-bit and 3.5inch Arduino Display-UNO.

Marvin121287, 2021-12-12:
Hi Moon,
I have a problem to get my Pi Pico to work with an 8-bit Arduino Display. I already got your library to work with the 2" Pico TFT with SPI and now want to try out an 8-bit tft.
I connect D0-D7 to GPIO 0-7 and CS to 17, RS to 27, WR to 18 and RD to 19. Of course GND to GND, VSYS to 5V and 3V3 to 3V3. 
The LCD background led lights up and I can upload code to Pi (tried Fade example from board library). I solderes every pin to eachother, so there is no lose connection and checkt everything multiple times to not get the wrong GPIO. Maybe I need the Reset pin to connect to a GPIO pin?

My screen ist that one: http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO so I used the ili9486 class for the screen and of course the 8-bit bus.

After upload, nothing happens and the screen just stays full white.

I used your hello world example with the class and bus fitted to my setup:

--------------------------------------------------

#include <Arduino_GFX_Library.h>

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_RPiPicoPAR8(27 /* DC */, 17 /* CS */, 18 /* WR */, 19 /* RD */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_ILI9486_18bit(bus, -1 /* RST */, 0 /* rotation */, false /* IPS */);

void setup(void)
{
    gfx->begin();
    gfx->fillScreen(BLACK);

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(5000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    delay(1000); // 1 second
}

--------------------------------------------------

Hope you can help me with my problem. Or maybe I just solder some wrong GPIOs to each other :D.

Greetings Marvin

Stauzebench, 2021-12-15:
Hi Moon,
first thanks for your time to help me.

I have tried your new class for the screen. The problem now is that I can see the sketch running if I tilt the screen and its like distorted in some kind.

[image]

I hope you see what I mean. Sadly I have no idea how to handle that behavior. In SPI I would maybe reduce the Clock, but i 8-bit I am clueless. 

I dont know if something is wrong with the wiring, but I tried out the ili9486_18bit class and its the same behavior, but you can see a bit more at the angle. Its the endscreen with the results.

[image]

Maybe you have an idea whats wrong.

moononournation, 2021-12-16:
I have changed to use @nopnop2002 init code, please see is it work.

Stauzebench, 2021-12-16:
We are making some progress. The display is now displaying, but in some weird kind of way. And it looks like its going mirrowed.

[image]

Thanks for going through this and implement another library just to make this work. You are great

nopnop2002, 2021-12-16:
@Stauzebench 

Your TFT may not be ILI9486.
You can use this to find out your driver.
https://github.com/prenticedavid/MCUFRIEND_kbv/tree/master/examples/LCD_ID_readreg

This is for Arduino UNO.
It is my ILI9486.
Check reg(0x00D3).

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Powsr Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 07 0F 29 0A 1B 0D 69 B3 37 04 07 07 0D 03 0A	GAMMA-P
reg(0x00E1) 00 08 3F 2B 05 0F 06 2E 35 26 0D 0E 03 2F 2E 0D	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control
```

Marvin121287, 2021-12-17:
Hey nopnop2002 and Moon,

the output of the readReg listed the following 

`Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 01 01 22 15 81 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 3B 00 02 00 01 00 43	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 03 03 24 04 00	NVM Read
reg(0x00D3) 03 03 24 04	ILI9341, ILI9488
reg(0x00D4) 03 03 24 04	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 07 07 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
`

I seems its an ili 9488 display. Hope you can do something to get this working now :)

moononournation, 2021-12-17:
you may get update and try Arduino_ILI9488 display class

nopnop2002, 2021-12-19:
@Marvin121287 

>I seems its an ili 9488 display. 

No.
This is my ILI9488.
Check reg(0x00D3).

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Powsr Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0	GAMMA
reg(0x00CC) 00 03	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 05	NVM Read
reg(0x00D3) 00 00 94 88	ILI9341, ILI9488
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 00 07 0C 05 13 09 36 AA 46 09 10 0D 1A 1E 0F	GAMMA-P
reg(0x00E1) 00 00 20 23 04 10 06 37 56 49 04 0C 0A 33 37 0F	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 58 04 12 02 22 22 FF 0A 90 14 88	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
```

Unfortunately your TFT driver cannot be identified.

Stauzebench, 2021-12-20:
Haha, I guess I just looked at the same regpoints and just assumed something because it stands there.
I know what you mean with buying displays. I bought some mentioned ili9481 and never got them running properly.
But thanks for enlighten me with your examples.

BTW this is my PCB of that screen: (Sorry for hiding the parts tha may be more interesting like the mentioned ILI9486 driver and the pinout lables).

[image]

I was searching the world wide web for a picture with that PCB but there is nothing with that solder point and the tracemarks of the bus.
I can remember that I got it running last year with an UNO and MEGA with the mentioned HX83.. driver I think, but I dont know how to implement that in that the library. If there would be a tutorial somewhere I would try it by myself instead of bothering someone else like you two.

Of course the new class setup was no success because of the wrong driver.

moononournation, 2021-12-20:
I suggest you use multimeter double check each connection from Pico to the display board 1 by 1.

nopnop2002, 2021-12-21:
@Stauzebench 

I think R61581 and ILI9481 have the same control code.

Doesn't it work with ILI9481?

Marvin121287, 2021-12-23:
I tried ILI9481, but it does not work. 
The problem with checking the wires is, that the chip connects to everything, so every pin beeps to eachother, I already tryed that out.
I think without the ability to try it out by yourself, its some kind of endless guessing game for that display. And even if you have that display, like nopnop said, every shield is different, thats kinda sad.
But I thank you for all the help with that screen and I already purchased some other displays to try out like the ILI9806, R61529 and an ST7793 from OPEN-SMART. And for realy sure a ILI9341 with 8-bit and SPI support.

BTW Moon, does your canvas behave like sprites from TFT_eSPI and Lovyan03?

nopnop2002, 2021-12-23:
@Marvin121287 

Thank you.

closed 2022-05-15 by moononournation

moononournation, 2022-10-11:
I found some incorrect interpretation of function in Arduino_ESP32PAR16 and now fixed. Please get update in github and test again.