moononournation/Arduino_GFX #108: ImgViewerAnimatedGIF/GifClass.h has memory bug?

marcmerlin, 2022-01-03:
I've tried to use this example code so that the gif is done in loop() and not setup()

As soon as I do, the code starts crashing with memory handling errors.
```
CORRUPT HEAP: Bad head at 0x3ffc1ddc. Expected 0xabba1234 got 0x3ffdfff8
Crash in
    void gd_close_gif(gd_GIF *gif)
    {
        gif->fd->close();
----->  free(gif->table);
        free(gif);
    }
```

Please look at this harmless diff that causes the corrupt HEAP
https://github.com/marcmerlin/FastLED_ArduinoGFX_TFT/commit/30ef83b42e4c6eae6c0d99178e8004e6ee2a9b95

You can look at this earlier diff where all I tried to do was to move the gif animation from setup to loop, I think I did it right, but you can check:
https://github.com/marcmerlin/FastLED_ArduinoGFX_TFT/commit/f7646f5569f4f7da7771ff1c78c93cf167c73f12

This happens when I run on an SSD1331 with kiss2.gif. The bug may not be triggered with another gif or different size.

If you prefer to offer ImgViewerAnimatedGIF where gif decoding happens in loop(), which is what I tried to do here, feel free to :)

moononournation, 2022-01-15:
I have committed the changes move to loop() and tested with latest Arduino IDE and Arduino-esp32 firmware. Cannot reproduce the memory bug, may be already fixed in latest firmware.

marcmerlin, 2022-01-30:
thank you. I just tested this version and do not have a crash anymore, so I agree that the buffer overrun is probably fixed, or the new layout of the code doesn't trigger it anymore, thank you ;)

closed 2022-01-30 by marcmerlin