moononournation/Arduino_GFX #109: Advertise solution for full FastLED compatible framebuffer

marcmerlin, 2022-01-05:
I read https://github.com/moononournation/Arduino_GFX and 
https://github.com/moononournation/Arduino_GFX/wiki/Canvas-Class

I understand the pros and cons of framebuffers, and as you may know, I have this framebuffer to rule them all:
https://marc.merlins.org/perso/arduino/post_2020-03-16_Framebuffer_GFX_-Choosing-between-its-3-2D-APIs_-FastLED-XY_-NeoMatrix_-and-LEDMatrix_-and-detail-of-its-many-supported-hardware-backends.html
This allows the exact same code to work on:
- FastLED Pixels (all of them)
- SmartMatrix RGBPanels
- rpi-rgb-panels on rPi for bigger panels
- TFTs

This fixes the need for a full color framebuffer (RGB888 for compatibility with FastLED, SmartMatrix, LEDMatrix, and rpi-rgb-panels), and my code runs on top of yours:
* http://marc.merlins.org/perso/arduino/post_2022-01-02_FastLED_ArduinoGFX_TFT-on-top-of-Framebuffer_GFX-for-SPI-TFTs-like-SSD1331-or-ILI9341.html
* https://github.com/marcmerlin/FastLED_ArduinoGFX_TFT

Are you willing to advertise this as a solution on top of your TFT code?

Demo: https://youtu.be/Ui__2O9qBac

moononournation, 2022-02-03:
I have added an MultipleAnimatedGIF example, please see if it can fit for your case:
https://github.com/moononournation/Arduino_GFX/tree/master/examples/MultipleDisplay/MultipleAnimatedGIF

marcmerlin, 2022-02-03:
I'm not sure what you are asking for :)
This issue was to ask if you could advertise my code that runs on top of yours as a way for people to have a fully featured 24bit canvas/framebuffer compatible with other libs and APIs in a way that the same framebuffer can even be rendered on both an LCD and an RGBPanel at the same time if needed, and more generally to allow running any code that actually requires a framebuffer that you can read back from (some code is fine with write only and works with your lib, other code needs to read from the framebuffer to move data, flip/mirror the display, or scroll in some ways that are not supported by the underlying TFT hardware).
I'm happy to provide a diff to your main README to explain pros and cons if you'd like

moononournation, 2022-02-09:
Hope you can consider it is hard to list out all examples that use a library in that library page. And the list is hard to maintain by the library code developer.

In contrast, a project should list out all dependent libraries and it is maintainable.

closed 2022-02-09 by moononournation

marcmerlin, 2022-02-10:
Thanks for the reply @moononournation . I see you point. I did list your library as a dependency in my library, but what I was referring to are people already using your library and looking for framebuffer support.  They will not easily find it on your pages and may not know it exists.

Would you be open to me sending you an edit to https://github.com/moononournation/Arduino_GFX/wiki/Canvas-Class
 ?