moononournation/Arduino_GFX #110: how to setup pin LCD NT35510 LCD 480x800 in arduino mega 2560?

Alis-Dev-idn, 2022-01-13:
I'm confused about this library, which pins are used for the NT35510 LCD 480x800 on the Arduino Mega 2560?

moononournation, 2022-01-13:
As you can see at README Tobe Support Dev Board, this item is on the wish list.

dapostol73, 2023-03-23:
Any update on this, been trying to get this to work with my Arduino Mega 2560 with no success.  Even if you just paste the code for the example video on [YouTube](https://youtu.be/C_1ASzUN3bg), that would be great.

moononournation, 2023-03-23:
not yet have this databus

moononournation, 2023-03-25:
You are using 8-bit or 16-bit parallel interface?

moononournation, 2023-03-25:
Just added a Arduino_SWPAR16 databus class, you may get update to test it.

dapostol73, 2023-03-25:
Here is the board itself, it support both but mine comes from the factory with the 16 Bit Parallel resistor soldered, so I assume I require an Arduino_SWPAR16 databus.

http://www.lcdwiki.com/4.0inch_Arduino_Display-Mega2560_NT35510

dapostol73, 2023-03-25:
Happy to test too...if you need to see any of the hardware information, I've uploaded the custom LCDWIKI_KBV libraries here for the this NT35510 if that helps.

https://github.com/dapostol73/WeatherStationInternal/tree/main/lib/

moononournation, 2023-03-26:
I have not tested with the new class, can you help to test it?

dapostol73, 2023-03-26:
I did some testing yesterday, I got to work with following DataBus

Arduino_DataBus *bus = new Arduino_SWPAR16(38, 40, 39, 43,
37, 36, 35, 34, 33, 32, 31, 30,
22, 23, 24, 25, 26, 27, 28, 29);

I did find the clear was a bit slow, not sure if this is because this bus is Software versus using a hardware version for the Mega

moononournation, 2023-03-26:
Thx for your information. Seems this display aligned all pins well to port C and A, it is possible to make Arduino_AVRPAR16 databus for it.

moononournation, 2023-03-27:
Just create the Arduino_AVRPAR16 databus, please get update and test it:
```
Arduino_DataBus *bus = new Arduino_AVRPAR8(38 /* DC */, 40 /* CS */, 39 /* WR */, 43 /* RD */, 3 /* PORT LOW */, 1 /* PORT HIGH */);
```

dapostol73, 2023-03-27:
Getting this error when I try and use Arduino_AVRPAR16 directly.

Arduino_DataBus *bus = new Arduino_AVRPAR16(38, 40, 39, 43, 3 /* PORT LOW */, 1 /* PORT HIGH */);

`Linking .pio\build\megaatmega2560\firmware.elf
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o: In function `Arduino_AVRPAR16::writeC8D16D16(unsigned char, unsigned int, unsigned int)':
<artificial>:(.text+0x1ff4): undefined reference to `Arduino_AVRPAR16::WRITE16(unsigned int)'
<artificial>:(.text+0x200a): undefined reference to `Arduino_AVRPAR16::WRITE16(unsigned int)'
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o: In function `Arduino_AVRPAR16::writeC8D16(unsigned char, unsigned int)':<artificial>:(.text+0x203e): undefined reference to `Arduino_AVRPAR16::WRITE16(unsigned int)'
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o: In function `Arduino_AVRPAR16::writePixels(unsigned int*, unsigned long)':
<artificial>:(.text+0x2076): undefined reference to `Arduino_AVRPAR16::WRITE16(unsigned int)'
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o: In function `Arduino_AVRPAR16::write16(unsigned int)':
<artificial>:(.text+0x208e): undefined reference to `Arduino_AVRPAR16::WRITE16(unsigned int)'
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o:<artificial>:(.text+0x20a6): more undefined references to `Arduino_AVRPAR16::WRITE16(unsigned int)' follow
C:\Users\dapos\AppData\Local\Temp\ccXTsnAC.ltrans0.ltrans.o:(.rodata+0xa3): undefined reference to `Arduino_AVRPAR16::writeC8D16D16Split(unsigned char, unsigned int, unsigned int)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\megaatmega2560\firmware.elf] Error 1`

moononournation, 2023-03-27:
fixed the error, please get update and try again:
```
Arduino_DataBus *bus = new Arduino_AVRPAR16(38 /* DC */, 40 /* CS */, 39 /* WR */, 43 /* RD */, 3 /* PORT LOW */, 1 /* PORT HIGH */);
```

dapostol73, 2023-03-28:
It compile now, but when I upload there is unfortunately no response of update on the screen, it just stays black.

`Arduino_DataBus *bus = new Arduino_AVRPAR16(38 /* DC */, 40 /* CS */, 39 /* WR */, 43 /* RD */, 3 /* PORT LOW */, 1 /* PORT HIGH */);
Arduino_GFX *gfx = new Arduino_NT35510(bus, 41, 0 /* rotation */);`

moononournation, 2023-03-28:
wild guess only, try alter port number to 1, 3

dapostol73, 2023-03-28:
No change

moononournation, 2023-03-28:
Ok, I will check the code tonight. And the SWPAR16 should be a little bit faster in latest version.

dapostol73, 2023-03-28:
Thanks, ya....I tried the SWPAR16 again to verify it was working, notice it was a lot faster.

moononournation, 2023-03-28:
Just checked the code and fixed somethings. I think it should work now.

dapostol73, 2023-03-29:
Success!!! I'll test now with my more complicated program 

[image]

moononournation, 2023-03-29:
Great! Can you help to prepare a PDQgraphicstest Youtube link? I will add it to the README.md. Thanks in advance.

dapostol73, 2023-03-29:
Sure what is required for that?

moononournation, 2023-03-29:
Just run the PDQgraphicstest.ino example in Arduino_GFX from start to end

dapostol73, 2023-03-29:
Looks like we have a problem, the test freezes here. Also the testPixel() is really slow VS other draws

[image]

dapostol73, 2023-03-29:
My project is already read running, so I am good, I just replaced my wrapper on LCDWIKI lib that make the same calls.

moononournation, 2023-03-29:
Ok, then I close this request.

closed 2023-03-29 by moononournation

dapostol73, 2023-03-29:
Should I open another issue about the crash/freeze when running PDQgraphictest.ino then?

moononournation, 2023-03-29:
If you encounter similar issue in your project, you can raise it. I think it just because high resolution of PDQgraphicstest pushed to touch the MEGA 2560 resource limit, e.g. out of memory.

dapostol73, 2023-03-30:
Here is the video of the PDQ Graphics Test running, not sure you want this public yet....but allows you to see how slow it is when doing some of the tests.

https://www.youtube.com/watch?v=10VSWyrh2gw

moononournation, 2023-03-30:
Thx, yes this is pushing the limit of the MEGA 2560. But I think this is already better than SWPAR16 a lot.

dapostol73, 2023-03-30:
Not sure about that, I did two tests.

1. Removed the 2 arc test and it completed fine.
2. Ran each Arc test on its own and they both frooze.

Think it something to do with the arc function. In my app I have my own, which is why I'm not seeing the freeze

moononournation, 2023-03-30:
Arc tests are the only test using floating point. It may be the hint

dapostol73, 2023-04-06:
So I did some more digging into the Arc Test.  It seems that the issue start as soon as the Inner\Outer Radius of the arc gets over 180.  I did some serial prints and it is still running....but for some reason it doesn't draw anything an become super sluggish.

For example, the first line works, the second and third one are slow and draw nothing.
```
  gfx->fillArc(240, 400, 174, 177, 0, 150, RED);
  gfx->fillArc(240, 400, 180, 183, 0, 150, RED);
  gfx->fillArc(240, 400, 186, 189, 0, 150, RED);
```

moononournation, 2023-04-06:
Thx for your findings. Seem it caused by the different behavior of handling floating point in different platform. The arc function is come from other libraries, may be I should rewrite it use integer only just like circle functions.