moononournation/Arduino_GFX #113: GC9A01 Display blinking at an interval of 1 second on ESP32S board

saturnhades, 2022-01-22:
I have tried all examples, but in each, the GC9A01 stars blink at an interval of 1 second. I have also tried both HWSPI and SWSPI but nothing changes. Solve this issue as this is so frustrating.

moononournation, 2022-01-22:
Most ESP32 problems are come from unstable power source. May be you can try disable WiFi at the beginning can help a little bit.

saturnhades, 2022-01-22:
Can you help me with a sample code for it

saturnhades, 2022-01-22:
I am having the same issue with the Arduino mega

saturnhades, 2022-01-22:
``/*******************************************************************************
 * Start of Arduino_GFX setting
 * 
 * Arduino_GFX try to find the settings depends on selected board in Arduino IDE
 * Or you can define the display dev kit not in the board list
 * Defalult pin list for non display dev kit:
 * Arduino Nano, Micro and more: TFT_CS:  9, TFT_DC:  8, TFT_RST:  7, TFT_BL:  6
 * ESP32 various dev board     : TFT_CS:  5, TFT_DC: 27, TFT_RST: 33, TFT_BL: 22
 * ESP32-C3 various dev board  : TFT_CS:  7, TFT_DC:  2, TFT_RST:  1, TFT_BL:  3
 * ESP32-S2 various dev board  : TFT_CS: 34, TFT_DC: 26, TFT_RST: 33, TFT_BL: 21
 * ESP8266 various dev board   : TFT_CS: 15, TFT_DC:  4, TFT_RST:  2, TFT_BL:  5
 * Raspberry Pi Pico dev board : TFT_CS: 17, TFT_DC: 27, TFT_RST: 26, TFT_BL: 28
 * RTL8720 BW16 old patch core : TFT_CS: 18, TFT_DC: 17, TFT_RST:  2, TFT_BL: 23
 * RTL8720_BW16 Official core  : TFT_CS:  9, TFT_DC:  8, TFT_RST:  6, TFT_BL:  3
 * RTL8722 dev board           : TFT_CS: 18, TFT_DC: 17, TFT_RST: 22, TFT_BL: 23
 * RTL8722_mini dev board      : TFT_CS: 12, TFT_DC: 14, TFT_RST: 15, TFT_BL: 13
 * Seeeduino XIAO dev board    : TFT_CS:  3, TFT_DC:  2, TFT_RST:  1, TFT_BL:  0
 * Teensy 4.1 dev board        : TFT_CS: 39, TFT_DC: 41, TFT_RST: 40, TFT_BL: 22
 ******************************************************************************/
#include <Arduino_GFX_Library.h>
# include <WiFi.h>

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
//Arduino_DataBus *bus = create_default_Arduino_DataBus();
Arduino_DataBus *bus = new Arduino_ESP32SPI(12 /* DC */, 5 /* CS */, 18 /* SCK */, 23 /* MOSI */, NULL /* MISO */, VSPI /* spi_num */);
//Arduino_DataBus *bus = new Arduino_ESP8266SPI(5 /* DC */, 15 /* CS */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, TFT_RST, 0 /* rotation */, false /* IPS */);
//Arduino_GFX *gfx = new Arduino_GC9A01(bus, 15 /* RST */, 0 /* rotation */, true /* IPS */);
Arduino_GFX *gfx = new Arduino_GC9A01(bus, 33 /* RST */, 0 /* rotation */, true /* IPS */);

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    WiFi.mode(WIFI_OFF);
    gfx->begin();
    /*delay(2000);
    Serial.begin(115200);
    Serial.println(MOSI);
    Serial.println(MISO);
    Serial.println(SCK);
    Serial.println(SS);*/
    gfx->fillScreen(BLACK);

#ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
#endif

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    //delay(5000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    //delay(1000); // 1 second
}``

This is the code i am using and after attaching the display the following error occurs:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
19:37:40.615 -> flash read err, 1000
19:37:40.615 -> ets_main.c 371 
19:37:40.940 -> ets Jun  8 2016 00:22:57

Also, the display blinks the same at a 1-second interval, and the code freezes. I am using the NodeMCU 32S Ai thinker model [](https://docs.ai-thinker.com/_media/esp32/docs/nodemcu-32s_product_specification.pdf) link to its datasheet. Also, the video link to demonstrate my problem is below https://drive.google.com/file/d/1-Yavphy9X2hjrdsdlo-MtxFN-EQUDqZa/view?usp=sharing. I hope you can find a solution for this soon.

saturnhades, 2022-01-23:
@moononournation could you tell me the problem or the reason of it

moononournation, 2022-01-26:
Wild guess try shorter wires.