moononournation/Arduino_GFX #117: FreeFonts are not working on ESP8266

Kotik82, 2022-01-24:
Hello, i was using ur library with GC9A01 on atmega328p in combination with free fonts and everything was working fine.

So i ported my project to a Wemos D1 Pro Mini (i need more memory to run a canvas) and everything is working fine BUT the fonts are not working anymore.

I can only use the library's default font and any other font is not working.

I tested with the Font example that u have and this one is not working either. On atmega328 everything was working fine. I dont get any errors when i compile or upload my code.

Below is my code:

#include <Arduino_GFX_Library.h>
#include "FreeSerif12pt7b.h"
#include <ESP8266WiFi.h>
#define TFT_BL 16

Arduino_DataBus *bus = new Arduino_ESP8266SPI(5 /* DC */, 2 /* CS */);
Arduino_GFX *gfx = new Arduino_GC9A01(bus, -1 /* RST */, 0 /* rotation */, true /* IPS */);

void setup() {
    WiFi.mode( WIFI_OFF );
    WiFi.forceSleepBegin();
    gfx->begin();
    gfx->fillScreen(BLACK);
    gfx->setFont(&FreeSerif12pt7b);
    #ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
    #endif
    delay(1000);                               // Delay
}

void loop() {
 
    gfx->setTextColor(WHITE);
    gfx->setCursor(70, 40);
    gfx->print("Current Temp: ");
 
}
This doesnt show any text, but when i switch to gfx->setFont( ); then everything is working BUT ofc with ugly fonts.

Kotik82, 2022-01-25:
After more troubleshooting i found out that the reason fonts are not working is this part in the fonts:

const uint8_t [] PROGMEM = {

Specifically if i remove PROGMEM then everything is working. Either the compiler is unable to upload to flash or the library is unable to read from flash, so by forcing the font into RAM and bypassing flash memory is a temporary solution. But i why is this happening? And why only on esp8266 and not on atmega?

moononournation, 2022-01-26:
Committed a fix, please help to test.

moononournation, 2022-01-26:
And FYI, you may find more font or tools for generate your own font at this repository:
https://github.com/moononournation/ArduinoFreeFontFile

Kotik82, 2022-01-26:
Hello, so i updated the library and re-downloaded fresh fonts from your font repository (fyi i was using your font repository for previous tests aswell), i did a fresh compile and i am still getting the same issue, nothing is displayed,.

And again if i replace const GFXglyph FreeMono8pt7bGlyphs[] PROGMEM = {   
to
const GFXglyph FreeMono8pt7bGlyphs[]  = {

then everything works :(

I did test uploading with "Erase flash: all flash contents" just to double check that i do everything right, still same issue.

Kotik82, 2022-01-26:
Ok i got more info. I did more testing with different parameters during upload/compile and i hope this helps u.

By default Arduino IDE has Non-32-bit Access: Use pgm_read macros for IRAM/PROGMEM 
if you change it to --> Byte/Word access to IRAM/PROGMEM 
then everything is working, maybe this will help you somehow.

https://ibb.co/YWcZ2Xv

Link to screenshot of the settings.

moononournation, 2022-02-03:
I have committed the fix by adding STORE_ATTR for ESP8266 and tested ok. Please help to verify.

lboross, 2022-02-20:
FYI, I have this exact same issue. Printing using free fonts crashed the ESP whenever trying to print.

Using the latest code and updated font header files is working for me.

Thanks for supporting this project.