moononournation/Arduino_GFX #122: Works fine with ESPboy (ESP8266) www.espboy.com

ESPboy-edu, 2022-02-07:
This beautiful library works fine with my ESPboy device
www.espboy.com
ESP8266 + ST7735 128x128

Please add ESPboy to the "Currently Supported Dev Device"

moononournation, 2022-02-08:
Welcome! But I do not have one in hand, can you give me a YouTube video demo link for PDQgraphictest?

moononournation, 2022-02-08:
and have you tried using Arduino_ESP8266SPI class?

ESPboy-edu, 2022-02-08:
Hello!
Here is the YouTube link https://youtu.be/Cx82XWrc8-0
Yes. With Arduino_ESP8266SPI its better.

```
#define SCAN_INTERVAL 3000
#define SPI_FREQUENCY 40000000

#define MCP23017address 0
#define CSTFTPIN     8 

#include <Arduino_GFX_Library.h>
#include <Adafruit_MCP23017.h> //to control buttons

Adafruit_MCP23017 mcp;

Arduino_DataBus *bus = new Arduino_ESP8266SPI(16/*TFT_DC*/, -1/*TFT_CS*/);
Arduino_GFX *gfx = new Arduino_ST7735(bus, -1 /* RST */, 0 /* rotation */, false /* IPS */,
    128 /* width */, 128 /* height */, 2 /* col offset 1 */, 3 /* row offset 1 */, 2 /* col offset 2 */, 1 /* row offset 2 */);

void setup(){
  mcp.begin(MCP23017address);
  delay(100);
  mcp.pinMode(CSTFTPIN, OUTPUT);
  mcp.digitalWrite(CSTFTPIN, LOW);

  gfx->begin(SPI_FREQUENCY);
}
```

moononournation, 2022-02-09:
Ok, update README and Wiki.

The result figures is how long the item run in milliseconds, so smaller is better.

closed 2022-02-09 by moononournation