moononournation/Arduino_GFX #129: Adding the Adafruit Bluefruit Feather LE support

rafael-sul, 2022-02-22:
I want to use my Adafruit Bluefruit board with a GC9A01 LCD screen but your library doesn't support the board. I was looking through the library and it shows that it supports the NRF52840 chip. Our board has the NRF52832 chip. 

Is there any way you can add the compatibility for my board?

moononournation, 2022-02-22:
Any compile error?

rafael-sul, 2022-02-22:
We are trying to use a developer board that is NOT supported by this library. We are using an Adafruit Bluefruit nRF52 Feather. When trying to use a similar data bus (NRFXSPI) that you have listed, we get this compile error message: 
[image]
Here is the "Tools" settings we are running:
[image]

Is there anyway to get our developer board "Adafruit Bluefruit nRF52 Feather" to work with this library?

moononournation, 2022-02-23:
If the board implemented Arduino generic SPI class, Arduino_GFX can support transparently. So I ask is it have any compile error?

rafael-sul, 2022-02-23:
We cannot compile using the Adafruit Bluefruit Feather. Is there a way we can add it to the databus library?

moononournation, 2022-02-23:
try HWSPI instead of NRF first

rafael-sul, 2022-02-24:
Hey! It works! Thank you this is what we needed! If we have any questions, can we contact you again? Thanks