moononournation/Arduino_GFX #130: JpegClass in ImgViewerJpeg has an error

SagarPunjabi, 2022-03-09:
Hi, I'm using a Adafruit nrf52832 Bluefruit feather with a GC9A01 as my display. When I run the ImgViewerJpeg example, I get this error in the draw method: 

`JpegClass.h:64:9: error: 'FS' has not been declared
   64 |         FS *fs, char *filename, JPEG_DRAW_CALLBACK *jpegDrawCallback, bool useBigEndian,`

I assume this has to do with the include statements at the top: 
[image]

I tried only using the <Seeed_FS.h> library but that resulted in this error:
 `JpegClass.h:82:9: error: reference to 'File' is ambiguous
   82 |         File f = fs->open(filename, FILE_READ);
`
Any advice on how to fix this would be appreciated, thanks!

moononournation, 2022-03-15:
I have just changed <FS.h> to <SD.h>, hope it can help.

SagarPunjabi, 2022-03-15:
Unfortunately, that did not work. I'm still get the same error:
[image]

[image]

moononournation, 2022-03-15:
just committed to avoid using FS class, please take a look.

SagarPunjabi, 2022-03-15:
No longer getting the FS undeclared error! I think that fixed it. However, There is a new issue, it seems like there is no .open method in the JPEGDEC library.

[image]

moononournation, 2022-03-15:
JPEGDEC library error outside this library scope, may be you can try modify JPEGDEC.h line 198 and 199. Replace `FS.h` with `SD.h`.

moononournation, 2022-03-16:
Yes, you can contribute on it.

SagarPunjabi, 2022-03-22:
Hey, I tried your suggestion but replacing ` FS.h` with `SD.h` in the JPEGDEC library didn't work. I also tried to implement the `open` method for `SD.h` but that didn't work either. Is there anyway you can help provide support to get this example to work with a SD card?

moononournation, 2022-03-22:
Sorry, I did not have Adafruit nrf52832 Bluefruit feather. It is a little bit over budget for me.

TripleT-circuitmaker, 2022-03-23:
Yes! I have same problems with my Arduino BLE 33! Please Help!

SagarPunjabi, 2022-03-25:
I managed to get it working. I will contribute the code soon, this should work on the nano as well. Hope it can help @moononournation @TripleT-circuitmaker

closed 2022-03-25 by SagarPunjabi

moononournation, 2022-03-25:
I just recalled actually it has an old version that should work before JPEGDEC implement FS.h:
https://github.com/moononournation/Arduino_GFX/commit/0bc64b6c246f9a95898d5c21b0fb30af7efb0938

moononournation, 2022-03-27:
Please get update and try is it work.

SagarPunjabi, 2022-03-27:
Hi, this update worked for me. I also just submitted a PR with what I did to get it to work. The PR isn't necessary now, since your update works but maybe it will be of some use. Thanks again for the support!

laoadam, 2022-09-17:
Hi,
my ImgViewerJpeg got error of: File System mount failed, how to fix please?
the ImgViewerBmp works well.
Thanks
Adam

moononournation, 2022-09-17:
the example header have some steps need to follow.