moononournation/Arduino_GFX #131: AmebaD RTL8722DM MINI compiler error with ili9431 dislpay

cold63, 2022-03-14:
I am using the AmebaD RTL8722DM MINI board with the ILI9341. I run the HelloWorld.ino example code  and get the following error.

```
Arduino_ILI9331.cpp: In member function 'virtual void Arduino_ILI9331::setRotation(uint8_t)':

C:\Users\dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.1\variants\rtl8722dm_mini/variant.h:57:39: error: expected unqualified-id before numeric constant

 #define SS                            10

                                       ^
```
But if I choose the bw16 board, the compiled  is successful and display ok.

moononournation, 2022-03-14:
add @xidameng in loop

xidameng, 2022-03-14:
@cold63 @moononournation  
Hi all, 
This library at version 1.1.8 is running fine with RTL8722DM_MINI, after upgrading to 1.1.9, then I start to see errors

@ v1.1.9, I got different error msg 
> HelloWorld:23:45: error: 'TFT_RST' was not declared in this scope

only if I re-define **TFT_RST** to some other pin, then will I see the error log mentioned above, I think it's a compatibility issue here

xidameng, 2022-03-14:
@cold63 This line is from version 1.1.8, you can use it as a reference 

https://github.com/moononournation/Arduino_GFX/blob/4eaa0c3b14012945a1a5057274990aa621cf49d5/examples/HelloWorld/HelloWorld.ino#L16

moononournation, 2022-03-15:
should be fixed now

cold63, 2022-03-16:
> should be fixed now

Yes,the problem has been fixed.

moononournation, 2022-03-16:
good to see that

closed 2022-03-16 by moononournation