moononournation/Arduino_GFX #135: Creating two "Canvas" and executing begin will stop the operation

LusterWong, 2022-03-21:
HI,
I use ESP32, VSPI interface. Here is the setup code.
```
static Arduino_DataBus *Bus = new Arduino_ESP32SPI(21, 5, 18, 23, 19, VSPI, false);
static Arduino_GFX *Output = new Arduino_ST7796(Bus, 15, 0, true);
Arduino_GFX *Gfx1 = new Arduino_Canvas(320, 48, Output);
Arduino_GFX *Gfx2 = new Arduino_Canvas(320, 48, Output);

void setup()
{
    Gfx1->begin(60000000);
    Gfx2->begin(60000000); 
    Serial.println( "The new program." ); <----Unable to run line
}
```
I wanted to use two "Conves" as frame buffers for LVGL, but when two canvases are built, the program gets stuck at **Gfx2-> BEGIN (60000000);**
Did I do something wrong there ?
Thx.

moononournation, 2022-03-21:
What is your use case? Not yet support 2 canvas share same display.

LusterWong, 2022-03-21:
> 

I want to use it with LVGL, now using LCD 320x480 resolution, refresh slowly.
Want to test if two buffers can improve flush speed.

closed 2022-03-27 by moononournation